FreeRTOS/FreeRTOS #2: How to submit new port

cui632, 2019-11-22:
Hi :

I 'd lke t submit a new platform porting to FREERTOS repo. but Got fail to commit. 
I tried to pull a new request, but there is no different branch to create. how to fix ?

Vincent

RichardBarry, 2019-11-22:
I’m not completely sure how the repo is configured but others have submitted pull requests.  Did you use a process similar to that described here: https://akrabat.com/the-beginners-guide-to-contributing-to-a-github-project/ ?

cui632, 2019-11-22:
[image]
Does it works ?

xiaoxiacui, 2019-11-22:
I have recreated a new acoount to submit, it stil gets same result ...

xiaoxiacui, 2019-11-22:
I know how to deal !  and have submit to a pull request successfully

yuhui-zheng, 2020-02-13:
Hi, this repo is now kind of ready to accept PR. Though we are working on defining a generic process, we should be able to evaluate your PR and merge. 

ps: Your alias looks familiar and I think we sorted things out. :) I'm drafting a tutorial page for your port/demo, and shall get a review request in email shortly.

closed 2020-02-13 by yuhui-zheng

xiaoxiacui, 2020-02-13:
Hi Yuhui:

 

Thank you for your efficient work. .

 

Please give me a notice kindly when my PR has been merged into mainstream and published out to download. 

 

The RISC-V platform shall be delay to port due to coronavirus outbreak.  I will notice you when we start at first time …

 

 

Best Regards,

Vincent Cui