FreeRTOS/FreeRTOS #10: \ missing in FreeRTOS-Plus-CLI documentation

chrfle, 2020-02-12:
I hope this is the correct place to report error in the html help.
On this page: https://www.freertos.org/FreeRTOS-Plus/FreeRTOS_Plus_CLI/FreeRTOS_Plus_CLI_IO_Interfacing_and_Task.html

The example are missing \ for newline, carriage return and backspace.. I.e. it says `'n'` instead of `'\n'`.

Here are two examples:
`if( cRxedChar == 'n' )`
`FreeRTOS_write( xConsole, "rn", strlen( "rn" );`

yuhui-zheng, 2020-02-13:
Thank you for reporting this issue. I have updated the html for [FreeRTOS+IO Integration](https://www.freertos.org/FreeRTOS-Plus/FreeRTOS_Plus_CLI/FreeRTOS_Plus_CLI_IO_Interfacing_and_Task.html). I'm not sure when the next website deployment is scheduled, but when it's done could see four lines being changed -- 

```
if( cRxedChar == '\n' )
FreeRTOS_write( xConsole, "\r\n", strlen( "\r\n" );
```

closed 2020-02-13 by yuhui-zheng

chrfle, 2020-02-13:
@yuhui-zheng is the html documentation available in a repository so I can create pull requests against it?

RichardBarry, 2020-02-13:
Unfortunately not at this time, although that would be a good idea.