FreeRTOS/FreeRTOS #21: Use a relative URL for the FreeRTOS-Kernel submodule.

chrisnc, 2020-02-26:
The `url` option for the FreeRTOS-Kernel submodule can just be `../FreeRTOS-Kernel.git` rather than the full github URL. This would help organizations that would like to mirror both of these repos without having the FreeRTOS-Kernel submodule always point at github, which is not desirable.

lundinc2, 2020-03-06:
@chrisnc Thanks for bringing this issue up. I have merged the PR that addresses this issue, and will mark this as closed.

yuhui-zheng, 2020-03-06:
The GitHub web UI does not seem to play well with relative URL — the link is dead that user cannot click through. (See the attached. Cloning works, it's just the UI.) I'm going to revert the change in PR #27. @chrisnc Thank you for the suggestion though. 

[image]

chrisnc, 2020-03-07:
Given that you have to navigate into the `FreeRTOS` directory before seeing this, it seems less severe to me? A (more) prominent link to FreeRTOS-Kernel in the README or description of this repo might be more generally useful. One of the links you added in the reverting PR is broken: https://github.com/FreeRTOS/FreeRTOS/commit/42c627b2b88cb3b487fea983d8b566a8bbae54fa#r37685464

chrisnc, 2021-04-06:
Can this be revisited/un-reverted? GitHub finally fixed the issue with links to submodules with relative URL paths in the web UI: https://github.community/t/support-linking-relative-urls-on-submodules/2078/21
Here's an example someone posted in that thread that originally did not work, but now does: https://github.com/apmorton/submoduledemo
Let me know if I should open a new issue.

muneebahmed10, 2021-07-21:
Hi @chrisnc,

I tried testing this in PR #661, and it was harder to clone from forks unless the submodule has been forked as well. Also, there was only one submodule when this issue was created, but there are about 20 now, not all under the FreeRTOS organization. Making just the kernel a relative submodule won't be all that useful, and neither would making all the FreeRTOS submodules into relative links, as there are still the submodules under AWS and those that are third party.

If you would like to use relative submodules for mirroring the repositories internally, we'd recommend:
1. Using `git submodule set-url` to set the custom URL.
2. Committing your changes to your internal repository.
3. Rebasing or merging your changes when updating from upstream, and keep an eye out for changes that overwrite the submodule URLs, such as with a pre-commit hook.

However, since there are a lot more submodules now, doing so might be more cumbersome for you than leaving them as is.

muneebahmed10, 2021-10-20:
Hi @chrisnc,

I will close this issue as our current number of submodules means that it is no longer practical to use relative submodules. You can try our suggestions in our [earlier post](https://github.com/FreeRTOS/FreeRTOS/issues/21#issuecomment-884483356) for your specific use case.

Please open a new issue if you have other feature requests or suggestions.

Thank you