FreeRTOS/FreeRTOS #32: [BUG] Invalid configASSERT() in MingW demo project

RichardBarry, 2020-03-10:
**Describe the bug**
A configASSERT() in the Eclipse/MingW demo project is now invalid and preventing the demo from executing.

**Target**
- Development board: PC, MingW, Eclipse
- Instruction Set Architecture: Windows x86
- IDE and version: Eclipse
- Toolchain and version: MingW

**Host**
- Host OS: Windows
- Version: 10

**To Reproduce**
- Follow [these ](https://www.freertos.org/FreeRTOS-Windows-Simulator-Emulator-for-Visual-Studio-and-Eclipse-MingW.html)instructions to build using MingW/Eclipse.
- Start executing, an assert is hit in the function that initialises the heap, before any tasks are created.

**Expected behavior**
There shouldn't be an assert in FreeRTOS V10.3.x code.

**Screenshots**
N/A

**Additional context**
To fix, remove [this](https://github.com/FreeRTOS/FreeRTOS/blob/V10.3.1/FreeRTOS/Demo/WIN32-MingW/main.c#L373) assert, which is invalid and should not be there.  You can then also remove the line above it (the call to vPortGetHeapStats()), and the xHeapStats and xZeroHeapStats variables which will then be defined but not used.

Spudmn, 2020-07-21:
This is fixed with commit 2df5eeef5763045c4c74ff0e2a4091b7d19bea89

lundinc2, 2020-11-11:
Closing as this issue is now fixed.

closed 2020-11-11 by lundinc2