FreeRTOS/FreeRTOS #33: [BUG] No prototypes for vApplicationGetXXXTaskMemory() callbacks

staalb, 2020-03-10:
According to [the docs](https://www.freertos.org/a00110.html) the application writer must provide two callback functions if `configSUPPORT_STATIC_ALLOCATION` is set to 1: `vApplicationGetIdleTaskMemory()` and `vApplicationGetTimerTaskMemory()`.

However, there is no header file available in the source code of FreeRTOS which provides proper prototypes for these functions. It is good practice to have a single function prototype shared by code implementing the function and code calling the function.

GCC emit warnings for this when compiled with `-Wmissing-prototypes`.

Right now there are 'external' declarations for these functions in `task.c` (e.g. [here](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/10bbbcf0b9bcab1869b3bac3987d32116f351718/tasks.c#L419)). Could these declarations be moved to e.g. `tasks.h`?

yuhui-zheng, 2020-03-11:
Leaving some comments potentially for further discussion. 

I agree it's a good practice to not emit warning when compiling with ```-Wmissing-prototypes```. I believe the original intention for not having function prototype in ```task.h``` or ```timers.h``` is that these interfaces are not FreeRTOS kernel APIs, that users are not supposed to call them in application code. These interfaces are more like extension points for user to customize kernel API behavior. (Similarly, the [hook functions](https://www.freertos.org/a00016.html) follow the same approach.)

RichardBarry, 2020-03-11:
Indeed that is (today at least) the intention and a pattern used wherever there is a callback provided by the user.  There is an extern declaration in the C file that calls the callback, but not in the header file as in most cases the function will not exist.  All the examples that use the callbacks have the prototypes locally, where the callback is implemented.

I can see there may be advantages to having the prototypes in the header file, although also maybe confusing as the functions are not provided by the kernel but by the application writer.  If they are moved to the header file they would need to use the same preprocessor guards to ensure they were not included when that feature was disabled in the FreeRTOSConfig.h settings, and include a clear comment to state they are not FreeRTOS functions but functions the application writer provides.

Given the above I'm inclined to close this as a bug as it is the intended behaviour, although not immediately to give you the opportunity to reply.

My answer above assumes the compiler warning you received was in your application code, where the callback functions are implemented.  If that assumption is not correct, and the compiler warning is emitted when compiling a kernel file, then please let me know.

staalb, 2020-03-11:
@RichardBarry, indeed this triggers a compiler warning in my application code. So technically this might be more a feature request than a bug report. Nevertheless, I think that its important that an application developer gets a warning if the function argument/return types are changed in FreeRTOS, potentially preventing nasty bugs in the future.

@yuhui-zheng,  I can see the argument why it's confusing if prototypes like these are part of `tasks.h`. I think these callbacks _are_ part of the FreeRTOS API, they only work in a backwards direction. It is still FreeRTOS who is dictating how I should interact with the FreeRTOS code. Maybe it might less confusing to put callbacks like these in a separate header (e.g. `appl_callbacks.h`)? Such an header file can have a clear comment stating that the functions declared by the file must be implemented by the application.

I personally doubt if it's necessary to put preprocessor guards around the prototypes. It's only a declaration, consuming no resources at all. Moreover, even if `configSUPPORT_STATIC_ALLOCATION` is set to 0, the application writer might still have this function implemented somewhere. Most of the time, I trust the linker not to include a function definition if it's not used (using `-ffunction-sections` and `--gc-sections`) instead of using preprocessor conditionals.

closed 2023-04-20 by aggarg