FreeRTOS/FreeRTOS #35: [BUG] return type of xAreSemaphoreTasksStillRunning?

spth, 2020-03-12:
According to FreeRTOS/Demo/Common/include/semtest.h the return type is BaseType_t, according to FreeRTOS/Demo/Common/Full/semtest.c it is portBASE_TYPE. Obviously, this makes compilation fail when the two types differ.

Both files should use the same type.

I ran into this issue when trying to compile the Cygnal port with current SDCC. Apparently, for the Cygnal port one of the types is signed char, while the other is unsigned char.

RichardBarry, 2020-03-12:
portBASE_TYPE is a legacy definition that was deprecated many many years ago in favour of BaseType_t - so the two should be the same - the fact [they are not](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/V10.3.1-kernel-only/portable/SDCC/Cygnal/portmacro.h#L60) may be a quirk of this being an 8-bit part and whether a char defaults to signed or unsigned differs between compilers and even the options passed to the compiler.  

Both the Cygnal port and demo are also very old and have been listed on the 'known issues' list for long time as we don't prioritise updating it.  So you have the double issue of the demo also [referencing legacy 'common' demo files in the FreeRTOS/Demo/Common/Full ](https://github.com/FreeRTOS/FreeRTOS/blob/master/FreeRTOS/Demo/Cygnal/Makefile#L44)directory which were not updated to use BaseType_t.

To sort this for you I think the first thing to do is update your build to reference files in the FreeRTOS/Demo/Common/Minimal instead of the FreeRTOS/Demo/Common/Full directory, and search and replace use of portBASE_TYPE to BaseType_t within the files local to the demo itself (files like main.c) - then you will be using types and files that are not deprecated and any issues you are still facing after that we can fix in the main line - so please report back.

spth, 2020-03-12:
If deprecated years ago, why is Common/include/serial.h still using portBASE_TYPE? Or is that file deprecated too?

spth, 2020-03-12:
I am no really familiar with FreeRTOS or Cygnal devices. I don't even have one.
I just had a quick look at the port out of curiosity. However, I am somewhat familiar with C and SDCC. The attached patch is sufficient to make the Cygnal port compile with current SDCC (i.e. revision 11556 or later - an SDCC bug making FreeRTOS compilation fail was fixed just today).

However, it does not link yet (the Makefile expects a few object files in different directories from where they are built), and of course I did no testing.
[sdcc-current.patch.gz](https://github.com/FreeRTOS/FreeRTOS/files/4325988/sdcc-current.patch.gz)

closed 2021-10-19 by muneebahmed10