FreeRTOS/FreeRTOS #42: [BUG] Renesas RSK64F Demo build errors (linker)

befree68, 2020-03-23:
**Describe the bug**
There are 33 errors when trying to build the RX600_RX64M_RSK_Renesas_e2studio demo with the latest E2 Studio IDE.

**Target**
- Development board: Renesas RSK+RX64M (R5F564MLLCDFC processor)
- Instruction Set Architecture: RXv2
- IDE and version: Renesas E2 Studio Version 3.0.1.09
- Toolchain and version: RX Compiler CC-RX V2.01.00

**Host**
- Host OS: Windows 10
- Version: 10.0.15063 Build 15063

**To Reproduce**
You do not need hardware to reproduce this issue. Simply download the latest E2 Studio IDE from Renesas website [Windows only] and import the RX64M demo as described on the FreeRTOS website (https://www.freertos.org/RX64M_RTOS_Renesas_GCC_e2studio.html). The initial build fails with 33 linker errors related: 2 related to undefined ref to sprint and 31 related to undefined refs to stream buffer symbols.

**Expected behavior**
Building the demo project should complete successfully without any user modifications required.

**Screenshots**
N/A

**Additional context**
Console output:
```
Description	Resource	Path	Location	Type
E0562310: Undefined external symbol "_xStreamBufferReceiveFromISR" referenced in ".\Source\Common_Demo_Source\Minimal\StreamBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferReceiveFromISR" referenced in ".\Source\Common_Demo_Source\Minimal\MessageBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferSend" referenced in ".\Source\Common_Demo_Source\Minimal\MessageBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferSend" referenced in ".\Source\Common_Demo_Source\Minimal\MessageBufferAMP.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferSend" referenced in ".\Source\Common_Demo_Source\Minimal\StreamBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
recipe for target 'RTOSDemo.abs' failed [Conflicting names] 	makefile	/RTOSDemo/HardwareDebug	line 119	C/C++ Problem
make: *** [RTOSDemo.abs] Error 1	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferIsEmpty" referenced in ".\Source\Common_Demo_Source\Minimal\MessageBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferBytesAvailable" referenced in ".\Source\Common_Demo_Source\Minimal\StreamBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferGenericCreate" referenced in ".\Source\Common_Demo_Source\Minimal\StreamBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferGenericCreate" referenced in ".\Source\Common_Demo_Source\Minimal\MessageBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferGenericCreate" referenced in ".\Source\Common_Demo_Source\Minimal\MessageBufferAMP.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferIsEmpty" referenced in ".\Source\Common_Demo_Source\Minimal\StreamBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferSendCompletedFromISR" referenced in ".\Source\Common_Demo_Source\Minimal\MessageBufferAMP.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_sprintf" referenced in ".\Source\Common_Demo_Source\Minimal\MessageBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_sprintf" referenced in ".\Source\Common_Demo_Source\Minimal\MessageBufferAMP.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferGenericCreate" referenced in ".\Source\Common_Demo_Source\Minimal\StreamBufferInterrupt.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferSendFromISR" referenced in ".\Source\Common_Demo_Source\Minimal\MessageBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferSendFromISR" referenced in ".\Source\Common_Demo_Source\Minimal\StreamBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferIsFull" referenced in ".\Source\Common_Demo_Source\Minimal\MessageBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferIsFull" referenced in ".\Source\Common_Demo_Source\Minimal\StreamBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferReceive" referenced in ".\Source\Common_Demo_Source\Minimal\MessageBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferReceive" referenced in ".\Source\Common_Demo_Source\Minimal\StreamBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferSendFromISR" referenced in ".\Source\Common_Demo_Source\Minimal\StreamBufferInterrupt.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferReceive" referenced in ".\Source\Common_Demo_Source\Minimal\MessageBufferAMP.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferSpacesAvailable" referenced in ".\Source\Common_Demo_Source\Minimal\StreamBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferReset" referenced in ".\Source\Common_Demo_Source\Minimal\MessageBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferReceive" referenced in ".\Source\Common_Demo_Source\Minimal\StreamBufferInterrupt.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferSpacesAvailable" referenced in ".\Source\Common_Demo_Source\Minimal\MessageBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_vStreamBufferDelete" referenced in ".\Source\Common_Demo_Source\Minimal\MessageBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_vStreamBufferDelete" referenced in ".\Source\Common_Demo_Source\Minimal\StreamBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferReset" referenced in ".\Source\Common_Demo_Source\Minimal\StreamBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
E0562310: Undefined external symbol "_xStreamBufferNextMessageLengthBytes" referenced in ".\Source\Common_Demo_Source\Minimal\MessageBufferDemo.obj"	RTOSDemo		 	C/C++ Problem
```

befree68, 2020-03-23:
https://forums.freertos.org/t/renesas-rsk64m-demo-build-errors-undefined-external-symbol-sprintf/9286

NoMaY-jp, 2020-09-04:
I think that this pull request can be closed because the issue is solved by commit https://github.com/FreeRTOS/FreeRTOS/commit/1ae6eda77aa5665d50a889f814736121e7f9960e

cobusve, 2020-09-28:
CLosing as per NoMaY-jp above

closed 2020-09-28 by cobusve