FreeRTOS/FreeRTOS #58: Feature Request: FreeRTOS+TCP should cache more than one DNS result IP address

gkwicker, 2020-04-21:
FreeRTOS+TCP currently [caches only a single DNS result IP address](https://github.com/FreeRTOS/FreeRTOS/blob/master/FreeRTOS-Plus/Source/FreeRTOS-Plus-TCP/FreeRTOS_DNS.c#L1125) when `ipconfigUSE_DNS_CACHE == 1`.  This doesn't work well with  load balanced services where DNS returns a list of addresses for use – see [AWS Route 53 FAQ](https://aws.amazon.com/route53/faqs/) as one example.  As described in that FAQ, _"IP addresses associated with load balancers can change at any time due to scaling up, scaling down, or software updates."_.  Because it can take time for DNS to update, sometimes the IP address cached by FreeRTOS+TCP will not be valid.  For most clients, this is not an issue.  For example in Linux, [`gethostbyname()`](http://man7.org/linux/man-pages/man3/gethostbyname.3.html) returns all of the IP addresses associated with the hostname in a `struct hostent` structure.  If a client can't connect to one of the IP addresses, it can simply retry with the next one in the list.  However, in FreeRTOS+TCP if the single address stored in the cache is not valid, the application will be unable to connect to the service [until the TTL expires](https://github.com/FreeRTOS/FreeRTOS/blob/master/FreeRTOS-Plus/Source/FreeRTOS-Plus-TCP/FreeRTOS_DNS.c#L1498) and a new DNS query is sent.

Example DNS lookup of load-balanced service – _FreeRTOS+TCP should allow the DNS result cache to be configured such that it contains multiple IP addresses per host, and applications should be able to cycle through these addresses if it can't connect to one of them._

```sh
$ nslookup xxxxxxxxxxxxxx-ats.iot.us-west-2.amazonaws.com
Server:		xxx.xxx.xxx.xxx
Address:	xxx.xxx.xxx.xxx#53

Non-authoritative answer:
Name:	xxxxxxxxxxxxxx-ats.iot.us-west-2.amazonaws.com
Address: 52.4x.xxx.xxx
Name:	xxxxxxxxxxxxxx-ats.iot.us-west-2.amazonaws.com
Address: 52.2x.xxx.xxx
Name:	xxxxxxxxxxxxxx-ats.iot.us-west-2.amazonaws.com
Address: 52.1x.xxx.xxx
Name:	xxxxxxxxxxxxxx-ats.iot.us-west-2.amazonaws.com
Address: 34.2xx.xxx.xxx
Name:	xxxxxxxxxxxxxx-ats.iot.us-west-2.amazonaws.com
Address: 52.3x.xx.xxx
Name:	xxxxxxxxxxxxxx-ats.iot.us-west-2.amazonaws.com
Address: 44.2xx.xx.xxx
```

gkwicker, 2020-05-06:
An implementation and test for this feature are in [amazon-freertos PR #1981](https://github.com/aws/amazon-freertos/pull/1981).

lundinc2, 2020-11-12:
I believe this is now in the FreeRTOS + TCP library. Am closing, please re-open if that is incorrect.

Thanks,

Carl

closed 2020-11-12 by lundinc2