FreeRTOS/FreeRTOS #68: TCP support heap_1.c scheme for one socket using REUSE_LISTEN_SOCKET or similar flag [DOC]

NetanelBollag, 2020-05-16:
**Is your feature request related to a problem? Please describe.**

Currently, there is no option to use heap_1 and the TCP stack since any closing socket is result in vEventGroupDelete and vPortFree. Since any socket can be closed, it makes using TCP stack with heap1 scheme impractical. 

**Describe the solution you would like.**
 support heap_1 with reusing the socket in the following conditions:

1. Only one connection supported, listening socket resources will be used for the client socket, same as current action with REUSE_LISTEN_SOCKET
2. The socket shouldn't be closed, only shutdown
3. After shutdown, calling listen() on the the socket will result in returning the socket to listening mode, allowing it to reconnect
4. Like any other FreeRTOS object, the socket can be created and not destroyed, while the maintaining the socket behavior of opening and closing.

**Additional context**
This will allow implementing a simple server with one peer-to-peer connection only, without full implementation of static socket allocation. It's can be add with a new flag (e.g. FREERTOS_SO_ALLOW_RECONNECT_LISTEN_SOCKET) to keep the new behavior separated from the old REUSE_LISTEN_SOCKET.

The need for this feature came from a project I'm currently working on, and I believe that it can be useful for other device who want to implement TCP connection with restricted memory management.

I'm willing to work on it myself, if the community support the idea.

rawalexe, 2023-04-11:
Thank you for your request we can consider your PR if you can provide one and thank you for your feature-request.

NetanelBollag, 2023-04-13:
Thank you for your response. I'm no longer working on the project that needed this feature, so I won't be able to invest time. From what I remember back on my research back then, it should be a pretty straightforward implementation. Thanks again, and good luck!