FreeRTOS/FreeRTOS #87: ICMP checksum offoloading [Inquiry]

e-sr, 2020-06-07:
I am tring to port the tcpip stack to the msp432E from ti. The emac peripheral can perform header checksum offloading thus i am using `ipconfigDRIVER_INCLUDED_TX_IP_CHECKSUM` set to 1.
My driver implementation works fine in the case of  ip header checksum offloading. However it does not work in the case of ICMP, pings are discarded beacuse of checksum error. Looking at the function  `prvProcessICMPEchoRequest`  in `FreeRTOS/FreeRTOS-Plus/Source/FreeRTOS-Plus-TCP/FreeRTOS_IP.c` i realized that the ICMP  checksum is calculated despite of the option   `ipconfigDRIVER_INCLUDED_TX_IP_CHECKSUM`.
Commenting out the checksum part in `prvProcessICMPEchoRequest` as follow 
```c
/* due to compiler warning "integer operation result is out of range" */
#if ipconfigDRIVER_INCLUDED_TX_IP_CHECKSUM
pxICMPHeader->usChecksum=0;
#else
usRequest = ( uint16_t ) ( ( uint16_t )ipICMP_ECHO_REQUEST << 8 );

if( pxICMPHeader->usChecksum >= FreeRTOS_htons( 0xFFFFu - usRequest ) )
{
	pxICMPHeader->usChecksum = ( uint16_t )
		( ( ( uint32_t ) pxICMPHeader->usChecksum ) +
			FreeRTOS_htons( usRequest + 1UL ) );
}
else
{
	pxICMPHeader->usChecksum = ( uint16_t )
		( ( ( uint32_t ) pxICMPHeader->usChecksum ) +
			FreeRTOS_htons( usRequest ) );
}
#endif
```
solve the problem. Is it possible  to add an ipconfig defines 
`ipconfigDRIVER_INCLUDED_TX_ICMP_CHECKSUM` to simplify porting?

RichardBarry, 2020-06-07:
Would ipconfigDRIVER_INCLUDED_TX_IP_CHECKSUM and the suggested ipconfigDRIVER_INCLUDED_TX_ICMP_CHECKSUM ever be set differently?

Secondary to my question - it would be FANTASTIC if you could create a pull request to upstream your driver once it was complete.  We would probably keep it in a 'third party' direct unless we had a way of testing and maintaining ourselves thereafter.  Repo is here https://github.com/FreeRTOS/FreeRTOS/tree/master/FreeRTOS-Plus/Source/FreeRTOS-Plus-TCP

e-sr, 2020-06-08:
> Would ipconfigDRIVER_INCLUDED_TX_IP_CHECKSUM and the suggested
> ipconfigDRIVER_INCLUDED_TX_ICMP_CHECKSUM ever be set differently?

I dont' think. The port driver implementation uses the low level emac peripheral driver from ti (called driverlib) and there I'dont see an option to exclude ICMP checksum. 

For the second question, YES of course. I will make a pull request in a couple of weeks.

htibosch, 2020-06-23:
> Would ipconfigDRIVER_INCLUDED_TX_IP_CHECKSUM and the suggested
> ipconfigDRIVER_INCLUDED_TX_ICMP_CHECKSUM ever be set differently?

You are right that some EMAC peripherals do not calculate ICMP protocol checksums, whereas they do calculate UDP and TCP checksums.

But in stead of introducing a new macro, it can be solved in the NetworkInterface. E.g. in [STM32Fxx/NetworkInterface.c](https://github.com/FreeRTOS/FreeRTOS/blob/master/FreeRTOS-Plus/Source/FreeRTOS-Plus-TCP/portable/NetworkInterface/STM32Fxx/NetworkInterface.c), around line 657, you see that the field is set to zero.

Could you do the same in the msp432E driver? Btw, Is that a public driver or can it be made public?

PS. see also my comments in #92 just now.

e-sr, 2020-07-01:
I made a comment on  #92. We can discuss it there.
I am woorking on a pullrequest for my driver

closed 2020-07-01 by e-sr