FreeRTOS/FreeRTOS #90: [BUG] PIC24_MPLAB Demo Fail#2 and crash

hp-beko, 2020-06-16:
**Describe the bug**
PIC24_MPLAB Demo Fail#2 and crash after some time.
The Demo on PIC24FJ128GA010

**Target**
- Development board: Explorer 16/32 Developement Board & PIC24F 100P TO 100P QFP PLUG IN MODULE
- Instruction Set Architecture: PIC24F
- IDE and version: MPLABX 5.15
- Toolchain and version: xc16 v1.41

**Host**
- Host OS: Windows
- Version: 10

**To Reproduce**
- Use project [RTOSDemo_PIC24.zip](https://github.com/FreeRTOS/FreeRTOS/files/4785783/RTOSDemo_PIC24.zip)
- Run on the Development Board and could observe wrong blinking LEDs and after some time the text "Fail#2" on display occured.

**Expected behavior**
The expected behavior should be like described on the [demo application page](https://www.freertos.org/portpic24_dspic.html#DemoApp)

RichardBarry, 2020-06-16:
The error reported [shows that the serial port communications has failed](https://github.com/FreeRTOS/FreeRTOS/blob/V10.3.1/FreeRTOS/Demo/PIC24_MPLAB/main.c#L220), so I'm not sure this is a bug yet.  It may have been better to discuss [on the support fourm](https://forums.freertos.org/) before opening a bug.

First question would be, did you have a loopback connector fitted as per the instructions you link to?  The loopback connector is required to ensure all transmitted characters are also received.

lundinc2, 2020-07-24:
Hello @hp-beko,

Do you have any update on this? I will close this issue shortly, as it has become stale.

Thank you

hp-beko, 2020-07-29:
Hello,

@RichardBarry yes, you are right! My software team didn't read the instructions correctly.
We are very sorry for this Issue!

closed 2020-07-29 by hp-beko