FreeRTOS/FreeRTOS #107: Unable to open mbedtls/check_config.h

mohitbalany, 2020-06-24:
I am trying to run the MQTT Demo with TLS (https://www.freertos.org/mqtt/preconfiguredexamplesTLS.html) on  Visual Studio 2019(FreeRTOS WIndows Port). However, when I build the solution, (mqtt_basic_tls_demo.sln) I get the following error:

Error (active)	E1696	cannot open source file "/mbedtls/check_config.h"		
Error	C1083	Cannot open include file: 'mbedtls_config.h': No such file or directory
Error	C1083	Cannot open include file: 'mbedtls/ctr_drbg.h': No such file or directory

I checked the folder mbedtls which was empty. I deleted the empty folder and cloned mbedtls ( from https://github.com/ARMmbed/mbedtls)

and included the path in additional directories from project properties. However, errors still persist.

Prior to this, I had successfully built and ran the solution file for the plain text demo (https://www.freertos.org/mqtt/preconfiguredexamples.html)

What else can I try doing?
Any suggestions would be highly appreciated! Thanks!

RichardBarry, 2020-06-24:
Sorry to hear you are having an issue.  Did you follow the instructions to clone the repo, so it brought in the submodules?  If you need to follow up please switch to using the support forums here https://forums.freertos.org - thanks.

closed 2020-06-24 by RichardBarry

mohitbalany, 2020-06-24:
Yes i followed the instructions I was able to remove those errors. However
my issues was taken off before i could edit it.

I bought in the submodules of mbedtls from the branch on the git. But a
file inside it by the name md_wrap.c throws error of " Too many initializer
values" (E1046, C2078).

Any way to fix this?

Let me know,
Regards
Mohit Balany