FreeRTOS/FreeRTOS #154: [Inquiry] A report of build check result using FreeRTOS V10.3.1 and CC-RX/GNURX/ICCRX for Renesas RX MCUs

NoMaY-jp, 2020-07-19:
Hello,

This is a report, not a question. After discussion the result with Renesas Electronics Corporation, I will consider what I can do.

Recently I noticed that Uncrustify breaks inline asm code for Renesas CC-RX compiler (https://github.com/FreeRTOS/FreeRTOS-Kernel/issues/90). Additionally, I noticed open issue (https://github.com/FreeRTOS/FreeRTOS/issues/42) and open pull request  (https://github.com/FreeRTOS/FreeRTOS/pull/43) regarding to a build problem of the RX600_RX64M_RSK_Renesas_e2studio RTOSDemo project. Today I was interested how many RTOSDemo projects for Renesas RX MCUs can be built with recent IDEs and recent compilers, so that I checked it. The result is as follows:

```
IDEs:

e2 studio v7.8.0
EWRX V4.14.1

Compilers:

CC-RX V3.02.00
GNURX 2020 q2 (8.3.0.202002)
ICCRX V4.14.1.5408 (included in EWRX V4.14.1)

Result:
                                        e2 studio   e2 studio   EWRX        
                                        CC-RX       GNURX       ICCRX       
RX100_RX113-RSK_GCC_e2studio_IAR        -           Fail        Pass        
RX100_RX113-RSK_Renesas_e2studio        Fail        -           -           
RX100-RSK_GCC_e2studio                  -           Fail        -           
RX100-RSK_IAR                           -           -           Pass        
RX100-RSK_Renesas_e2studio              Fail(*)     -           -           
RX200_RX210-RSK_Renesas                 N/A(HEW)    -           -           
RX200_RX231-RSK_GCC_e2studio_IAR        -           Fail        Fail        
RX200_RX231-RSK_Renesas_e2studio        Pass        -           -           
RX600_RX62N-RDK_GNURX                   -           N/A(HEW)    -           
RX600_RX62N-RDK_IAR                     -           -           Pass        
RX600_RX62N-RDK_Renesas                 N/A(HEW)    -           -           
RX600_RX62N-RSK_GNURX                   -           N/A(HEW)    -           
RX600_RX62N-RSK_IAR                     -           -           Pass        
RX600_RX62N-RSK_Renesas                 N/A(HEW)    -           -           
RX600_RX63N-RDK_Renesas                 N/A(HEW)    -           -           
RX600_RX630-RSK_Renesas                 N/A(HEW)    -           -           
RX600_RX64M_RSK_GCC_e2studio            -           Fail        -           
RX600_RX64M_RSK_Renesas_e2studio        Fail        -           -           
RX700_RX71M_RSK_GCC_e2studio_IAR        -           Fail        Pass        
RX700_RX71M_RSK_Renesas_e2studio        Pass        -           -           

Note:

'Pass' : The project can be built. (Note that only build check is passed.)
'Fail' : The project cannot be built. (i.e. compiler error(s) or linker error(s) are caused.)
'Fail(*)' : The project file cannot be migrated to the latest project file format. (i.e. I cannot start build.)
'N/A(HEW)' : The project is for HEW (another IDE) which is no longer as important as it was.
'-' : The project is not for the compiler.

All projects of Pass/Fail needed to be migrated to the latest project file format.
Additionally, when using e2 studio, toolchain type and toolchain version needed to be re-selected.
```

lundinc2, 2020-07-24:
Hello @NoMaY-jp,

Thank you for the detailed report. We have begun investigating this issue, and will update in this thread once more progress has been made.

NoMaY-jp, 2020-07-29:
Thank you for your attention @lundinc2. I have new information. The following projects can be built using Renesas's another IDE 'CS+' which is popular in Japan but which is released not only in Japan but also worldwide. (On the other hand, HEW is outdated.) CS+ can convert HEW's workspace/project file such as 'RTOSDemo.hws'/'RTOSDemo.hwp' to CS+'s native project file such as 'RTOSDemo.mtpj' and moreover CS+ can generate an intermediate project file such as 'RTOSDemo.rcpe' and e2 studio can convert such intermediate project file to e2 studio's native project file '.project' and '.cproject'. By using such way, the following projects can be built using e2 studio now on my desktop.
```
Build check result:
                                        e2 studio v2020-07      CS+ V8.04       e2 studio v2020-07(via CS+/.rcpe)
                                        CC-RX                   CC-RX           CC-RX
RX200_RX210-RSK_Renesas                 N/A(HEW)                Pass            Pass
RX600_RX62N-RDK_Renesas                 N/A(HEW)                Pass            Pass
RX600_RX62N-RSK_Renesas                 N/A(HEW)                Pass            Pass
RX600_RX63N-RDK_Renesas                 N/A(HEW)                Pass            Pass
RX600_RX630-RSK_Renesas                 N/A(HEW)                Pass            Pass
```

NoMaY-jp, 2020-08-10:
Hello @lundinc2 and @alfred2g,

This week I'd like to start to fix build errors of RTOSDemo projects for Renesas RX MCUs because I'm finishing to fix build errors of RTOSDemo projects for Renesas RL78 MCUs in the following thread.

[https://forums.freertos.org/t/compilation-linker-issues-in-running-the-demo-example-for-freertos-demo-rl78-rl78g13-promo-board-iar-directory/9218/7](https://forums.freertos.org/t/compilation-linker-issues-in-running-the-demo-example-for-freertos-demo-rl78-rl78g13-promo-board-iar-directory/9218/7)

Best regards,
NoMaY

lundinc2, 2020-08-10:
Hi @NoMaY-jp,

Thank you for doing this. In parallel we will be working on resolving the build issues related to uncrustify, so hopefully there will not be too many merge conflicts.

NoMaY-jp, 2020-08-20:
Hello,

I'm trying to fix build errors. I'll divide my tasks into mainly three steps and its several sub steps as follows.

**Step1) Fix build erros with FreeRTOS v10.3.1 but programs which are built aren't checked.**

1-1) Fix CC-RX projects then a zip file including fixed files will be uploaded to FreeRTOS forum.
1-2) Fix ICCRX projects then a zip file including fixed files will be uploaded to FreeRTOS forum.
1-3) Fix GNURX projects then a zip file including fixed files will be uploaded to FreeRTOS forum.

**Step2) Check projects using something else in stead of their target evaluation boards because I don't have them.**

2-1) Check RX71M and RX64M projects (RX600v2 port) but using RX65N evaluation board (also RX600v2 port).
2-2) Check RX231 projects (RX600v2 port) using other RX231 evaluation board (also RX600v2 port).
2-3) Check RX113 and RX111 projects (RX100 port) but using RX130 evaluation board (also RX100 port).
2-4) Check RX210 project (RX200 port) using other RX210 evaluation board (RX200 port).
2-5) Check RX630 project (RX600 port) using other RX630 evaluation board (RX600 port).
2-X) Regarding to RX62N-RDK, RX62N-RSK and RX63N-RDK (these are all RX600 port), I have no idea how to check these projects as of today. Moreover these projects use Ethernet and uIP task.

When mistake will be found, zip files which are uploaded to FreeRTOS forum will be updated.

**Step3) Post pull request(s).**

Best regards,
NoMaY

NoMaY-jp, 2020-09-04:
Hello,

My plan was changed  because Amazon may accept pull requests without checking programs working if there are no changes in the runtime functionality.

Before:

Step1) Fix build errors with FreeRTOS v10.3.1 but programs which are built aren't checked.
Step2) Check projects using something else in stead of their target evaluation boards because I don't have them.
Step3) Post pull request(s).

Now:

**Other than HEW projects**

Step1-A) Fix build errors with recent FreeRTOS code but programs which are built aren't checked to work.
Step1-B) Post pull requests for each fix.

Step2-A) Upgrade project files from older e2 studio to recent e2 studio and update them to use recent CC-RX and GNURX.
Step2-B) Check programs using something else in stead of their target evaluation boards because I don't have them.
Step2-C) Add CS+ project files for CC-RX.
Step2-D) Post pull requests for each upgrade.

Exception) RX200_RX231-RSK_GCC_e2studio_IAR's IAR project was checked to work using TB-RX231 evaluation board here https://github.com/FreeRTOS/FreeRTOS/pull/239.

Aditionally 'Style: Uncrustify' will be posted regarding to FreeRTOS-kernel's RX100 and RX600v2 ports.

**HEW projects**

Step3-A) Add project files for recent e2 studio (CC-RX, GNURX) and CS+ (CC-RX) which build the same MOT files as HEW.
Step3-B) Perhaps one pull request for all HEW projects will be posted.

Aditionally 'Style: Uncrustify' will be posted regarding to FreeRTOS-kernel's RX200 and RX600 ports.

As of today, the table is changed as follows:
```
IDEs:

e2 studio v7.8.0
EWRX V4.14.1

Compilers:

CC-RX V3.02.00
GNURX 2020 q2 (8.3.0.202002)
ICCRX V4.14.1.5408 (included in EWRX V4.14.1)

Status:
                                        e2 studio           e2 studio           EWRX
                                        CC-RX               GNURX               ICCRX
RX100_RX113-RSK_GCC_e2studio_IAR        -                   Fail                Pass
RX100_RX113-RSK_Renesas_e2studio        Fail-->Pass         -                   -
RX100-RSK_GCC_e2studio                  -                   Fail                -
RX100-RSK_IAR                           -                   -                   Pass
RX100-RSK_Renesas_e2studio              Fail(*)-->Pass      -                   -
RX200_RX210-RSK_Renesas                 N/A(HEW)            -                   -
RX200_RX231-RSK_GCC_e2studio_IAR        -                   Fail                Fail-->Pass(*)
RX200_RX231-RSK_Renesas_e2studio        Pass                -                   -
RX600_RX62N-RDK_GNURX                   -                   N/A(HEW)            -
RX600_RX62N-RDK_IAR                     -                   -                   Pass
RX600_RX62N-RDK_Renesas                 N/A(HEW)            -                   -
RX600_RX62N-RSK_GNURX                   -                   N/A(HEW)            -
RX600_RX62N-RSK_IAR                     -                   -                   Pass
RX600_RX62N-RSK_Renesas                 N/A(HEW)            -                   -
RX600_RX63N-RDK_Renesas                 N/A(HEW)            -                   -
RX600_RX630-RSK_Renesas                 N/A(HEW)            -                   -
RX600_RX64M_RSK_GCC_e2studio            -                   Fail                -
RX600_RX64M_RSK_Renesas_e2studio        Fail-->Pass         -                   -
RX700_RX71M_RSK_GCC_e2studio_IAR        -                   Fail                Pass
RX700_RX71M_RSK_Renesas_e2studio        Pass                -                   -

Note:

'Pass' : The project can be built. (Note that only build check is passed.)
'Fail' : The project cannot be built. (i.e. compiler error(s) or linker error(s) are caused.)
'Fail(*)' : The project file cannot be migrated to the latest project file format. (i.e. I cannot start build.)
'Pass(*)' : The program which is built is checked to work using other evaluation board.
'N/A(HEW)' : The project is for HEW (another IDE) which is no longer as important as it was.
'-' : The project is not for the compiler.

All projects of Pass/Fail needed to be migrated to the latest project file format.
Additionally, when using e2 studio, toolchain type and toolchain version needed to be re-selected.
```
Best regards,
NoMaY

NoMaY-jp, 2020-09-07:
Hello,

I notice (to be exact, I remember) that e2 studio doesn't upgrade projects for GNURX correctly. This is the reason why RTOSDemo projects are failed to be built in case of using recent versions of e2 studio. It's mandatory that upgraded projects are modified by hand. So, I'll post pull requests to fix incorrectly upgraded projects.

(On the other hand, e2 studio upgrades projects for CC-RX correctly so that upgrade project are able to be built without modification. And also EWRX upgrades its projects correctly other than one exception which isn't a standard project.)

Best regards,
NoMaY

dan4thewin, 2021-09-24:
Thank you for the PRs.  It appears the issue is concluded, so I'll close this.  If more attention is needed, please open a new issue.  Thank you.

closed 2021-09-24 by dan4thewin