FreeRTOS/FreeRTOS #179: A report of build check result using FreeRTOS V10.3.1 and ICCRX for Renesas RL78 MCUs

NoMaY-jp, 2020-07-29:
Hello,

This is a report, not a question. I'd like to post a pull request when I can fix it.

I was interested how many RTOSDemo projects for Renesas RL78 MCUs can be built with recent IDE and recent compiler because the following build problem was reported. So, I checked it. The result is the following.

Compilation/Linker Issues in Running the Demo Example for FreeRTOS/Demo/RL78_RL78G13_Promo_Board_IAR directory
[https://forums.freertos.org/t/compilation-linker-issues-in-running-the-demo-example-for-freertos-demo-rl78-rl78g13-promo-board-iar-directory/9218](https://forums.freertos.org/t/compilation-linker-issues-in-running-the-demo-example-for-freertos-demo-rl78-rl78g13-promo-board-iar-directory/9218)
```
IDE:

EWRL78 V4.20.1

Compiler:

ICCRL78 V4.20.1.2260 (included in EWRL78 V4.20.1)

Result:
                                        EWRX        
                                        ICCRX       
RL78_multiple_IAR                       Fail        
RL78_RL78G13_Promo_Board_IAR            Fail        
```

NoMaY-jp, 2020-08-18:
Hello,

Fixed source code is uploaded in the following post. Next step of this issue is to post a pull request.

Compilation/Linker Issues in Running the Demo Example for FreeRTOS/Demo/RL78_RL78G13_Promo_Board_IAR directory
[https://forums.freertos.org/t/compilation-linker-issues-in-running-the-demo-example-for-freertos-demo-rl78-rl78g13-promo-board-iar-directory/9218/10](https://forums.freertos.org/t/compilation-linker-issues-in-running-the-demo-example-for-freertos-demo-rl78-rl78g13-promo-board-iar-directory/9218/10)
FreeRTOSv10.3.1.EWRL78.Fixed.20200817.zip

Best regards,
NoMaY

dan4thewin, 2021-09-24:
Thank you for the report and the fixed source code.  I trust we can close this issue.  If more attention is needed, please open a new issue.  Thank you.

closed 2021-09-24 by dan4thewin