FreeRTOS/FreeRTOS #191: FreeRTOS plus submodules

crazycarl, 2020-08-07:
Would it be practical for the FreeRTOS plus libraries (like FreeRTOS+TCP) to be made submodules like the kernel?

I have been working with Rust bindings to the kernel, and the fact that the kernel is a submodule has made my task a lot easier.
I intend to write Rust bindings to FreeRTOS+TCP. If the same submodule approach were done for FreeRTOS+TCP, that would make this much easier as well.

RichardBarry, 2020-08-07:
This is great - would like to learn more.  To answer your question - yes - our intent is to make the TCP stack a sub module in the coming weeks.

IamTheCarl, 2020-08-07:
Oh that's perfect!

Well, I can't take credit for the current FreeRTOS bindings, but I have successfully used them in a toy project here.
I also did a lot of experiments with fixed point numbers in Rust. I'm trying to make the easiest embedded development platform I can. So far this is going far better than I expected.

Here's that toy project, which is very much still a work in progress, and may never actually get finished.
https://github.com/IamTheCarl/LocationTracker

I'll most likely be using the FreeRTOS+TCP on a different project though, since this board doesn't have the hardware needed.

Should this issue be left open until the submodules are made?

cobusve, 2020-10-27:
@IamTheCarl  -  FreeRTOS+TCP is now in it's own repository and submoduled into this repository just like you asked so I am closing this issue.

closed 2020-10-27 by cobusve

IamTheCarl, 2020-10-27:
Thank you very much.