FreeRTOS/FreeRTOS #244: [BUG]

ziyushijie, 2020-09-05:
**Describe the bug**
A concise description of what the bug is.

**Target**
- Development board: [e.g. HiFive11 RevB]
- Instruction Set Architecture: [e.g. RV32IMAC]
- IDE and version: [e.g. Freedom Studio 4.12.0.2019-08-2]
- Toolchain and version: [e.g. riscv64-unknown-elf-gcc-8.3.0-2019.08.0]

**Host**
- Host OS: [e.g. MacOS]
- Version: [e.g. Mojave 10.14.6]

**To Reproduce**
- Use project ... and configure with ...
- Run on ... and could observe ... 

**Expected behavior**
A concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
e.g. code snippet to reproduce the issue.
e.g. stack trace, memory dump, debugger log, and many etc.

ziyushijie, 2020-09-05:
When I use MDK's AC6 compiler, my whole project doesn't work. Port port uses files under GCC. I don't know if the poem file is a problem. If I use the port file under MDK, there are many errors when compiling. So, if I use the AC6 compiler of MDK, what should I do? thank you!

RichardBarry, 2020-09-05:
Sorry but I am confused by this bug report as the report talks about RISC-V and FreedomStudio and the but the additional comment about AC6 and MDK (which I think are for Arm).  I recommend posting to the support forum to see if we can help you there, and then if we determine there is a bug somewhere, we can open a bug ticket.

RichardBarry, 2020-09-05:
.....should have linked to the support forum, which is here: https://forums.freertos.org

closed 2020-09-05 by RichardBarry