FreeRTOS/FreeRTOS #273: Adding board support for shakti class of processors

sathya281, 2020-09-16:
Hi,

We want to add board support to Shakti class of processors. We want to upstream to the FreeRTOS official repository.

I want to know the possibilities of doing it ?
Also, the steps to do it. Please help here.

Background:
The [SHAKTI](http://shakti.org.in/) Processor Program, was started as an academic initiative back in 2014 by the RISE group at IIT-Madras. Source code of all the components of the SHAKTI ecosystem are open under the 3 part BSD license.  

We have a Hackathon kind of competition happening all over India. We have given an option to use FreeRTOS for the participants. We have ported the FreeRTOS on our boards. So, We feel up-streaming should help everyone of us.

Local ported repository is [here](https://gitlab.com/shaktiproject/software/FreeRTOS)

cobusve, 2020-09-28:
The way to do this is submit a port for the processor to the FreeRTOS-kernel repo via a pull request and then add the demo to this repository via a pull request.

RichardBarry, 2020-09-28:
As per Cobus' message above, we would like to help you upstream the code.  Looking at the PR above, which appears to be closed now, please:

- Ensure to use a future proof name for the directory the demo project uses.  The previous PR used FreeRTOS/Demo/shakti - please make the name of the processor more specific unless the demo will run on all shakti processors today and in the future, and also provide more detail in the directory name like which compiler and hardware the demo runs on.  Use the names of all the other FreeRTOS/Demo/* directories for reference.

- Do not check in object code.

- Is the port using our official RISC-V FreeRTOS port?  If so I'm not sure the clint driver code is necessary, although I could be wrong.  If the port is not using our official RISC-V port then I would like a discussion on what would be necessary to make it do so (missing features you need, or whatever, as it is best to get these into a single RISC-V port).

- How is the port tested?  The main() functions in the PR do not seem to do that much - normally we would create a simple demo  that is easy to understand for beginners (which these main() implementations may do), and then a comprehensive test/demo that builds a lot of the soak tests implemented in FreeRTOS/Demo/Common/Minimal.  Here is an example from a different RISC-V demo, although this one has very limited RAM so doesn't create many of the soak tests:
https://github.com/FreeRTOS/FreeRTOS/blob/master/FreeRTOS/Demo/RISC-V_RV32_SiFive_HiFive1-RevB_FreedomStudio/main.c#L117
https://github.com/FreeRTOS/FreeRTOS/blob/master/FreeRTOS/Demo/RISC-V_RV32_SiFive_HiFive1-RevB_FreedomStudio/full_demo/main_full.c#L146

nsathya123, 2020-12-28:
Dear Richard,

 Thanks for this. The pull request is [this](https://github.com/FreeRTOS/FreeRTOS/pull/299) 
I will address your comments and get back to you.

nsathya123, 2020-12-28:
Ensure to use a future proof name for the directory the demo project uses. The previous PR used FreeRTOS/Demo/shakti - please make the name of the processor more specific unless the demo will run on all shakti processors today and in the future, and also provide more detail in the directory name like which compiler and hardware the demo runs on. Use the names of all the other FreeRTOS/Demo/* directories for reference.

<response>
We have three boards Vajra, Parashu and Pinaka. All the three share the same BSP code, but capabilities are different.
Detail website is here: https://shakti.org.in/sp2020-shakti.html

Documentation - https://shakti.org.in/docs/shakti-soc-user-manual.pdf
		            https://shakti.org.in/docs/shakti-soc-device-register-manual.pdf
		            https://shakti.org.in/docs/boot_manual.pdf

    Do not check in object code.

<response> Yes, we wont.

    Is the port using our official RISC-V FreeRTOS port? If so I'm not sure the clint driver code is necessary, although I could be wrong. If the port is not using our official RISC-V port then I would like a discussion on what would be necessary to make it do so (missing features you need, or whatever, as it is best to get these into a single RISC-V port). 

<response>
We are using the official port and removing the clint code. We will remove unnecessary code.

    How is the port tested? The main() functions in the PR do not seem to do that much - normally we would create a simple demo that is easy to understand for beginners (which these main() implementations may do), and then a comprehensive test/demo that builds a lot of the soak tests implemented in FreeRTOS/Demo/Common/Minimal. Here is an example from a different RISC-V demo, although this one has very limited RAM so doesn't create many of the soak tests:
    https://github.com/FreeRTOS/FreeRTOS/blob/master/FreeRTOS/Demo/RISC-V_RV32_SiFive_HiFive1-RevB_FreedomStudio/main.c#L117
    https://github.com/FreeRTOS/FreeRTOS/blob/master/FreeRTOS/Demo/RISC-V_RV32_SiFive_HiFive1-RevB_FreedomStudio/full_demo/main_full.c#L146

<response>
Actually we have a demo for each SoC. They are below,

Demo directories are,
 1. shakti/vajra/ - Demo for vajra SoC.
 2. shakti/pinaka/ - Demo for pinaka SoC
 3. shakti/parashu/ - Demo for Parashu SoC

The documentation for realizing this is there at the beginning. We are also adding a readme.md for simple reading.

aggarg, 2021-10-04:
Closing this Issue as this has been merged in the community supported demos - FreeRTOS/FreeRTOS-Community-Supported-Demos#2

closed 2021-10-04 by aggarg