FreeRTOS/FreeRTOS #276: [DOC]https demo return 400 error

xjtrab, 2020-09-17:
**Describe the issue**
https demo return 400 error
**Reference**
https://www.freertos.org/https/preconfiguredexamples.html
**Screenshot**
If applicable, please attach screenshot.

**Browser**
- Browser: vs 2019  console

**i just change democonfigENABLE_TLS to 1**

console notes:
The following network interfaces are available:

Interface 1 - rpcap://\Device\NPF_{4B9FC3C4-4218-45FA-9656-586F022C8B0E}
              (Network adapter 'Intel(R) Ethernet Connection (7) I219-LM' on local host)

The interface that will be opened is set by "configNETWORK_INTERFACE_TO_USE", which
should be defined in FreeRTOSConfig.h
Attempting to open interface number 1.
Successfully opened interface number 1.
0 86 [MAC_ISR] Bounced back: 00:11:11:11:11:41
1 546 [IP-task] prvAllowIPPacket: UDP packet from 10.10.10.1 without CRC dropped
2 5326 [MAC_ISR] Bounced back: 00:11:11:11:11:41
3 5846 [IP-task] prvAllowIPPacket: UDP packet from 10.10.10.1 without CRC dropped
4 15566 [MAC_ISR] Bounced back: 00:11:11:11:11:41
5 16066 [IP-task] prvAllowIPPacket: UDP packet from 10.10.10.1 without CRC dropped
6 32386 [IP-task] prvAllowIPPacket: UDP packet from 10.10.10.1 without CRC dropped
7 35806 [MAC_ISR] Bounced back: 00:11:11:11:11:41
8 36306 [IP-task] prvAllowIPPacket: UDP packet from 10.10.10.1 without CRC dropped
9 76046 [MAC_ISR] Bounced back: 00:11:11:11:11:41
10 156276 [IP-task] ---------STARTING DEMO---------
11 156276 [IP-task]

IP Address: 10.10.10.200
12 156276 [IP-task] Subnet Mask: 255.0.0.0
13 156276 [IP-task] Gateway Address: 10.10.10.1
14 156276 [IP-task] DNS Server Address: 208.67.222.222

15 156276 [HTTPSDemo] Attempt to connect to httpbin.org:443
16 156346 [MAC_ISR] Bounced back: 00:11:11:11:11:41
17 156906 [MAC_ISR] Bounced back: 00:11:11:11:11:41
18 161386 [HTTPSDemo] DNS[0x727A]: The answer to 'httpbin.org' (3.221.81.55) will be stored
19 161546 [HTTPSDemo] Connected to httpbin.org:443
20 161546 [HTTPSDemo] Sending HTTPS GET request...
21 161687 [HTTPSDemo] Response status: 400
22 161687 [HTTPSDemo] Response body:
<html>
<head><title>400 The plain HTTP request was sent to HTTPS port</title></head>
<body bgcolor="white">
<center><h1>400 Bad Request</h1></center>
<center>The plain HTTP request was sent to HTTPS port</center>
</body>
</html>

23 161687 [HTTPSDemo]

24 161687 [HTTPSDemo] [WARN ][NET][161687] (Network connection 01032190) Failed to close connection.
25 161687 [HTTPSDemo] [WARN ][HTTPS Client][161687] Failed to shutdown the socket with error code: 4
26 161687 [HTTPSDemo] Disconnected from httpbin.org:443

27 161687 [HTTPSDemo] prvHTTPSDemoTask() completed an iteration successfully. Total free heap is 2079024
28 161687 [HTTPSDemo] Demo completed successfully.
29 161687 [HTTPSDemo] Short delay before starting the next iteration....

30 166687 [HTTPSDemo] Attempt to connect to httpbin.org:443
31 166726 [MAC_ISR] Bounced back: 00:11:11:11:11:41
32 167346 [HTTPSDemo] Connected to httpbin.org:443
33 167346 [HTTPSDemo] Sending HTTPS GET request...

xjtrab, 2020-09-17:
respHandle->pBody : 

&lt;html&gt;
&lt;head&gt;&lt;title&gt;400 The plain HTTP request was sent to HTTPS port&lt;/title&gt;&lt;/head&gt;
&lt;body bgcolor=&quot;white&quot;&gt;
&lt;center&gt;&lt;h1&gt;400 Bad Request&lt;/h1&gt;&lt;/center&gt;
&lt;center&gt;The plain HTTP request was sent to HTTPS port&lt;/center&gt;
&lt;/body&gt;
&lt;/html&gt;

lundinc2, 2020-11-12:
Hi  @xjtrab, you will want to use the TLS projected mentioned in https://www.freertos.org/https/preconfiguredexamplesTLS.html to use TLS. Please let me know if this resolves your issue.

lundinc2, 2020-11-24:
Hi @xjtrab I am closing this issue as it has gone stale. Please reopen it should you need further assistance.

Thanks,

Carl

closed 2020-11-24 by lundinc2