FreeRTOS/FreeRTOS #288: [BUG] Increasing clock speed changes led blink frequency

sprucify, 2020-09-22:
When I change `configCPU_CLOCK_HZ` from 10000000 to 20000000 in the Atmega4809 demo the led blinks slower.
I was expecting that clk_config.h and configTICK_RATE_HZ should keep the led blink frequency the same.

Note: I'm running this on a custom board with a Atmega4808

RichardBarry, 2020-09-22:
Are you also changing the CPU frequency to 20000000, or just the value of the configTICK_RATE_HZ constant?  This is probably better as a support request until we have concluded it is a bug, in which case we could write up how to replicate and therefore enable us to fix it.

sprucify, 2020-09-22:
I only changed `configCPU_CLOCK_HZ`
The file `clk_config.h` sets the right clock divide register.
And the rest should be calculated automatically with `configTICK_RATE_HZ` and `configCPU_CLOCK_HZ`

RichardBarry, 2020-09-22:
What is configUSE_TIMER_INSTANCE set to?  https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/master/portable/GCC/AVR_Mega0/porthardware.h#L18  Can you look at the calculations in Tick_Init() to see if there are any over or underflows.

sprucify, 2020-09-23:
I am on instance 0 which is a 16 bit Timer B.
`TCB0.CCMP = configCPU_CLOCK_HZ / configTICK_RATE_HZ;` will be 20000000 / 1000 = 20000

sprucify, 2020-09-23:
[image]

When I set `#define configCPU_CLOCK_HZ 20000000` it somehow managed to protective write 0x11 to register `MCLKCTRLB`

This means it sets the MCU clock divide to 6 = 3.33 Mhz

sprucify, 2020-09-23:
I fixed it by setting `MCLKCTRLB` to 0

`#if ( configCPU_CLOCK_HZ == 20000000 )

    #define CLK_init()  { \
						_PROTECTED_WRITE(CLKCTRL.MCLKCTRLA, CLKCTRL_CLKSEL_OSC20M_gc); \
						_PROTECTED_WRITE(CLKCTRL.MCLKCTRLB, 0); \
						}`

Can I do a pull request for it?

RichardBarry, 2020-09-24:
Sure, please do.

cobusve, 2020-09-28:
This was fixed in PR #295

closed 2020-09-28 by cobusve