FreeRTOS/FreeRTOS #314: [BUG] FreeRTOS+TCP v2.2.2 does not compile when ipconfigUSE_TCP is set to 0

jjr-simiatec, 2020-10-02:
`FreeRTOS_Sockets.c` and `FreeRTOS_TCP_WIN.c` do not compile when `ipconfigUSE_TCP` is disabled in `FreeRTOSIPConfig.h`.

[BuildLog.txt](https://github.com/FreeRTOS/FreeRTOS/files/5318549/BuildLog.txt)

**Suggested fixes**

Add `#if( ipconfigUSE_TCP == 1 )`/`#endif` guards as follows:

`FreeRTOS_Sockets.c`: declaration and implementation of `prvSockopt_so_buffer()`.

`FreeRTOS_TCP_WIN.c`: entire source.

Patches for the fixes as described above are attached.

[FreeRTOS_TCP_WIN.c.patch.txt](https://github.com/FreeRTOS/FreeRTOS/files/5318440/FreeRTOS_TCP_WIN.c.patch.txt)
[FreeRTOS_Sockets.c.patch.txt](https://github.com/FreeRTOS/FreeRTOS/files/5318441/FreeRTOS_Sockets.c.patch.txt)

cobusve, 2020-10-27:
Hi John, thanks for reporting this. We will check this out shortly. If you want to make a contribution yourself you are welcome to submit that patch as a PR to the repository. 

FreeRTOS+TCP has been moved to its own repo at https://github.com/FreeRTOS/FreeRTOS-Plus-TCP to make it easier to manage, if you feel like creating the PR yourself so that it will show your contribution then you should do it over there.

AniruddhaKanhere, 2020-11-06:
Hello @jjr-simiatec, I am closing this issue since a fix for it has been merged into the mainline.
You are welcome to re-open this issue or create a new one if something else is amiss.

Thanks again for catching the issue!

closed 2020-11-06 by AniruddhaKanhere