FreeRTOS/FreeRTOS #347: [BUG] Old Trace Recorder in FreeRTOS-Plus

eriktamlin, 2020-10-15:
**Describe the bug**
I have a pull request at https://github.com/FreeRTOS/FreeRTOS/pull/330 for the latest Trace Recorder.
Is this how I should draw attention to a pull request for an updated recorder so it makes it into coming releases?
Or just post in the forums? Which is better?

**Target**
N/A

**Host**
N/A

**To Reproduce**
N/A

**Expected behavior**
N/A

**Screenshots**
N/A

**Additional context**
N/A

cobusve, 2020-10-27:
Hi Erik, 

We have been working through a list of PR's to test as part of the upcoming release and it has been going slower than we would like. We got to #330 and are in the process of testing if it all working. I sent you a message in that PR about updating it with Master, but I am afraid as soon as you have done that it will go out of date again, so the easiest method is to give access for your fork to be updated, then we can just hit the button in github to do that. 

This works well enough to get attention to a PR, we do get to them all in the end. If you need to get immediate attention on something having a discission on the forum should get attention faster.

cobusve, 2020-10-29:
The PR has been merged

closed 2020-10-29 by cobusve