FreeRTOS/FreeRTOS #432: why taskpool lib malloc memeory using static and dynamic both?

RogerYaozhenyu, 2020-12-01:
at iot_threads_freeftos.c   please see the function as below.  i am confonfused that , if i only want to use dynamic . it' could not link susscessful. 

``
bool Iot_CreateDetachedThread( IotThreadRoutine_t threadRoutine,
                               void * pArgument,
                               int32_t priority,
                               size_t stackSize )
{
    bool status = true;

    configASSERT( threadRoutine != NULL );

    IotLogDebug( "Creating new thread." );
    threadInfo_t * pThreadInfo = IotThreads_Malloc( sizeof( threadInfo_t ) );

    if( pThreadInfo == NULL )
    {
        IotLogDebug( "Unable to allocate memory for threadRoutine %p.", threadRoutine );
        status = false;
    }

    /* Create the FreeRTOS task that will run the thread. */
    if( status )
    {
        pThreadInfo->threadRoutine = threadRoutine;
        pThreadInfo->pArgument = pArgument;

        if( xTaskCreate( _threadRoutineWrapper,
                         "iot_thread",
                         ( configSTACK_DEPTH_TYPE ) stackSize,
                         pThreadInfo,
                         priority,
                         NULL ) != pdPASS )
        {
            /* Task creation failed. */
            IotLogWarn( "Failed to create thread." );
            IotThreads_Free( pThreadInfo );
            status = false;
        }
    }

    return status;
}

bool IotMutex_Create( IotMutex_t * pNewMutex,
                      bool recursive )
{
    _IotSystemMutex_t * internalMutex = ( _IotSystemMutex_t * ) pNewMutex;

    configASSERT( internalMutex != NULL );

    IotLogDebug( "Creating new mutex %p.", pNewMutex );

    if( recursive )
    {
        ( void ) xSemaphoreCreateRecursiveMutexStatic( &internalMutex->xMutex );
    }
    else
    {
        ( void ) xSemaphoreCreateMutexStatic( &internalMutex->xMutex );
    }

    /* remember the type of mutex */
    if( recursive )
    {
        internalMutex->recursive = pdTRUE;
    }
    else
    {
        internalMutex->recursive = pdFALSE;
    }

    return true;
}

yourslab, 2020-12-03:
That is just the nature of the code - The taskpool requires memory to be allocated both statically and dynamically. However, the taskpool is being deprecated in our coming-soon LTS release by rewriting our libraries to not depend on it.

As for the linking error, could you please provide the error so we can help you debug? Perhaps, you must define both of these macros in `FreeRTOSConfig.h`:
```
#define configSUPPORT_DYNAMIC_ALLOCATION        1
#define configSUPPORT_STATIC_ALLOCATION         1
```

RogerYaozhenyu, 2020-12-04:
yes, it coule be linked success, after i defined configSUPPORT_STATIC_ALLOCATION   1  at FreeRTOSConfig.h . i just confused that this is not flow the freertos memory  guidance.  it seems that taskpool does not support dynamic memory mangment.
BTW.   if taskpool is being deprecated . will other library  provide the same feature? thanks.

yourslab, 2020-12-04:
Other libraries will provide the same features, but tasks will be allocated by the application. This way, all memory management will be up to the user. If you need a mechanism for creating tasks in your application, I suggest taking a [look here](https://www.freertos.org/a00019.html).

RichardBarry, 2020-12-04:
Grateful if support questions can use https://forums.freertos.org - thanks.

closed 2020-12-04 by RichardBarry