FreeRTOS/FreeRTOS #527: Integrating espressif ULP assmbler files with Amazon FreeRTOS

rupans, 2021-03-17:
Hi,

I am currently trying to build and link my assembler files within ESP32 vendor code within the Amazon FreeRTOS platform.
I have followed the instructions by espressif and in the .rst files within Amazon FreeRTOS. However when I try to compile I get the following error logs:

-- Building for target esp32
CMake Error at vendors/espressif/esp-idf/components/ulp/component_ulp_common.cmake:55 (add_dependencies):
  add_dependencies called with incorrect number of arguments
Call Stack (most recent call first):
  vendors/espressif/boards/esp32/CMakeLists.txt:397 (include)
  CMakeLists.txt:70 (include)

CMake Error at vendors/espressif/esp-idf/components/ulp/component_ulp_common.cmake:58 (target_add_binary_data):
  target_add_binary_data Function invoked with incorrect arguments for
  function named: target_add_binary_data
Call Stack (most recent call first):
  vendors/espressif/boards/esp32/CMakeLists.txt:397 (include)
  CMakeLists.txt:70 (include)

The relevant code in the CMakeLists.txt file is as follows and i Have highlighted the error line below:

set(ULP_APP_NAME ulp_${COMPONENT_NAME})
set(ULP_S_SOURCES "${board_dir}/application_code/ulp/ulp_adc.S")
set(ULP_EXP_DEP_SRCS "${board_dir}/application_code/ulp_adc_example_main.c ${board_dir}/application_code/params.c")
include(${esp_idf_dir}/components/ulp/component_ulp_common.cmake)    # <------- this line is the issue

The section of code from the component_ulp_common.cmake file is as follows:

add_dependencies(${COMPONENT_TARGET} ${ULP_APP_NAME}_artifacts)     
target_linker_script(${COMPONENT_TARGET} ${CMAKE_CURRENT_BINARY_DIR}/${ULP_APP_NAME}/${ULP_APP_NAME}.ld)
target_add_binary_data(${COMPONENT_TARGET} ${CMAKE_CURRENT_BINARY_DIR}/${ULP_APP_NAME}/${ULP_APP_NAME}.bin BINARY)
endif()

rupans, 2021-03-17:
I have also tried a solution with idf_component_register() and ulp_embed_binary() but when i do I get an unknown CMake command. Any ideas?

ravibhagavandas, 2021-03-20:
Hi rupans@

Could you provide following details?

1. version of Amazon FreeRTOS repository used.
2. version of ESP-IDF used
3. Are you trying to build Amazon FreeRTOS as a subdirectory within an ESP IDF project?

rupans, 2021-03-22:
Hi @ravibhagavandas,

Thank you for getting back to me. The answers to the questions are as follows:

1. version of Amazon FreeRTOS repository used.

    202007.00 July 2020 
    commit 36e1faf0b931dd8c78e32fc79df1a3f1718cb0c7 (HEAD -> master, origin/master, origin/HEAD)
    Merge: c20c8f274 a35a407cc
    Author: Ravishankar Bhagavandas <bhagavar@amazon.com>
    Date:   Wed Oct 7 16:40:42 2020 -0700

2. Version of ESP-IDF:
 
    commit 29889b575e9f6f72aada23ab24d82d8d177498f0 (HEAD, origin/release/v3.3, origin/HEAD, release/v3.3)
    Author: Shubham Kulkarni <shubham.kulkarni@espressif.com>
    Date:   Mon Jun 8 14:29:00 2020 +0530

3. I am building my code within Amazon FreeRTOS in the following location:
    <amazon-freeRTOS>/vendors/espressif/boards/esp32/...

Thank you again for you help.

Kind Regards,

Rupan

lundinc2, 2021-03-24:
Hi Rupan,

I see this is duplicated here https://forums.freertos.org/t/integrating-modern-esp32-ulp-toolchain-with-aws-freertos/11822/45. I am posting this link for those who stumble on this thread, and will reply to you there once I have investigated.

I am going to close this issue, and we may continue on the forum.

Thanks,

Carl

closed 2021-03-24 by lundinc2