FreeRTOS/FreeRTOS #548: [Feature Request] Improvement to UARTCommandConsole.c

RichardBarry, 2021-04-04:
**Is your feature request related to a problem? Please describe.**
[UARTCommandConsole.c](https://github.com/FreeRTOS/FreeRTOS/blob/V10.4.1/FreeRTOS-Plus/Demo/Common/FreeRTOS_Plus_CLI_Demos/UARTCommandConsole.c) is a template that demonstrates how to use a UART as the IO interface for FreeRTOS+CLI.  The file has two variables with the same name.  First a file scope static intended to allow tasks other than the CLI task access the serial console.  The second is function scope variable for use in the FreeRTOS+CLI task itself.  The file scope variable is never changed from its initial value of 0, and its definition is hidden by the variable of the same name in FreeRTOS+CLI task.

**Describe the solution you'd like**
Remove the function scope variable so all tasks, including the FreeRTOS+CLI task, use the file scope variable.

See the forum post here: https://forums.freertos.org/t/uninitialized-use-of-static-variable-in-rtosdemo-examples-uartcommandconsole-c/12181

bharsaklemukesh975, 2023-04-27:
Can I work on this?

archigup, 2023-09-21:
> Can I work on this?

Hey, apologies for late reply. Feel free to work on the above fix; PRs are welcome!

dougwfoster, 2023-12-31:
I didn't see any updates for this, so I created a PR.

PR #1147 (2024-01-08): Improvement to UARTCommandConsole.c

Description
-----------
Removed the function scope **xPort** variable within FreeRTOS+CLI task in UARTCommandConsole.c in order for the task to use the static file scope **xPort** variable. Also, added a prototype of vOutputString.

Test Steps
-----------
Tested by running the FreeRTOS+CLI demo on MSP432

Checklist:
----------
- [] I have tested my changes. No regression in existing tests.
- [] I have modified and/or added unit-tests to cover the code changes in this Pull Request.

Related Issue
-----------
Fixes #548 

By submitting this pull request, I co [... cut]
closed 2024-01-08 by aggarg