FreeRTOS/FreeRTOS #565: Possible Error in serial.c at AVR_ATMega323_WinAVR Demo [BUG]

victorfdezc, 2021-04-17:
Hello,
When building the AVR_ATMega323_WinAVR Demo with "make", poisoned signal errors raise.
Specifically, these errors are the following:
`serial/serial.c:188:9: error: attempt to use poisoned "SIG_UART_RECV"
 SIGNAL( SIG_UART_RECV )
         ^
serial/serial.c:207:9: error: attempt to use poisoned "SIG_UART_DATA"
 SIGNAL( SIG_UART_DATA )
`
To solve the problem, I simply changed both signal names as [this documentation](http://www.nongnu.org/avr-libc/user-manual/group__avr__interrupts.html) suggests.
That is, `SIG_UART_RECV` is replaced by `USART_RXC_vect` and `SIG_UART_DATA` by `USART_UDRE_vect`.

I'm new to these kind of projects, so I hope this solves the problem successfully.

AniruddhaKanhere, 2021-04-18:
Hello @victorfdezc, thank you for bringing this issue to our attention! We will look into this issue and report back shortly. Thanks.

AniruddhaKanhere, 2021-04-19:
Hello @victorfdezcm,
I tried compiling the source with the latest IAR embedded workbench but did not get any errors. Can you let me know which version of the workbench are you using? And whether you are using some special flags of have some options while compiling the source?

victorfdezc, 2021-04-20:
Hello @AniruddhaKanhere, sorry for the late response.
I'm not using any IAR Embedded Workbench but avr-gcc compiler version 5.4.0. Also, I didn't modified the makefile, so I suppose I'm not using any special flags for compiling.

paulbartell, 2021-04-24:
@victorfdezc If you submit a pull request with your changes, we would be happy to take a look and verify the change.

yanjos-dev, 2021-06-30:
I'm closing this issue since a fix has been merged. Please open this issue if there are any other relevant concerns.

closed 2021-06-30 by yanjos-dev