FreeRTOS/FreeRTOS #572: [BUG] hardfault after xTimerStart in a task in NRF52832

jinxiangwzh, 2021-04-22:
``
static void sample_task_entry(void * pvParameter)
{
    UNUSED_PARAMETER(pvParameter);
    EventBits_t event;

   check_err_timer = xTimerCreate("Timer", 2000, pdFALSE,(void *) 0, check_error_timer_callback);
    if( check_err_timer == NULL )
    {
        NRF_LOG_ERROR("Timer create failure\r\n");
    }

    while (true)
    {
        event = xEventGroupWaitBits(
            sample_task_event_handle,
            IMU_DATA_DRDY_EVENT,
            pdTRUE,  /* bit should be cleared before returning. */
            pdFALSE, /* Don't wait for both bits, either bit will do. */
            portMAX_DELAY);
        if(event == IMU_DATA_DRDY_EVENT)
        {
            xTimerStart(check_err_timer, 0);
            if(icm4x6xx_read_raw_data(icm42686_dev) != SENSOR_SUCCESS)
            {
                NRF_LOG_ERROR("imu data read failure\r\n");
                NRF_LOG_PROCESS();
            }
        }
    }
}

HardFault Detail:
<error> hardfault: HARD FAULT at 0x000075B6

<error> hardfault:   R0:  0x0000002E  R1:  0x00004AB0  R2:  0xD1015F00  R3:  0xF0014601

<error> hardfault:   R12: 0x00000002  LR:  0x00008449  PSR: 0x41000000

<error> hardfault: Cause: Data bus error (PC value stacked for the exception return points to the instruction that caused the fault)

paulbartell, 2021-04-22:
@jinxiangwzh Can you provide a bit more detail about your code and a stack trace when the hardfault occurs?

paulbartell, 2021-04-28:
@jinxiangwzh Were you able to address this issue?

jinxiangwzh, 2021-06-18:
Sorry ,I can't  the issue is still exist

yanjos-dev, 2021-06-30:
Hi @jinxiangwzh, are you still having this issue? If so, can you please provide more information about the code and stack trace when the hardfault occurs?

gshvang, 2021-09-30:
Hi @jinxiangwzh,

Since we don't have enough information, closing this issue. Please reopen if you need additional help.

closed 2021-09-30 by gshvang