FreeRTOS/FreeRTOS #606: [DOC] xTaskResumeFromISR and maybe other api examples are not updated with correct portYIELD_FROM_ISR usage

Hadatko, 2021-05-26:
**Describe the issue**
At example at bottom of page https://www.freertos.org/taskresumefromisr.html is code
```
if( xYieldRequired == pdTRUE )
     {
         // We should switch context so the ISR returns to a different task.
         // NOTE:  How this is done depends on the port you are using.  Check
         // the documentation and examples for your port.
         portYIELD_FROM_ISR();
     }
```

and i think it should be:
```
 // We should switch context so the ISR returns to a different task.
 // NOTE:  How this is done depends on the port you are using.  Check
 // the documentation and examples for your port.
 portYIELD_FROM_ISR(xYieldRequired);
```

**Reference**
https://www.freertos.org/taskresumefromisr.html 

**Screenshot**
If applicable, please attach screenshot.

**Browser**
- Browser: Chrome
- Version: 90.xxx

RichardBarry, 2021-05-26:
Agreed - the required syntax depends on the port but only very old ports use the code sequence highlighted on this page so the documentation should be updated as suggested.

robhath, 2021-06-01:
This page has been updated. Thanks for pointing this out!

dachalco, 2021-06-01:
Thank you @robhath.

closed 2021-06-01 by dachalco

Hadatko, 2021-06-02:
Hi, thank you to solve this issue. Although i see a documentation a bit inconsistent. For example here: https://www.freertos.org/FreeRTOS-timers-xTimerStartFromISR.html
you ma see:
```
if( xHigherPriorityTaskWoken != pdFALSE )
    {
        /* Call the interrupt safe yield function here (actual function
        depends on the FreeRTOS port being used). */
    }
```