FreeRTOS/FreeRTOS #627: [Feature Request] <replace with your title>

jinxiangwzh, 2021-06-18:
Why is the code so messy? Why not use the style like [RT-Thread](https://github.com/RT-Thread) or linux

gshvang, 2021-06-18:
@jinxiangwzh It would be helpful for us if you can share where exactly you are finding the code messy.

dan4thewin, 2021-06-24:
One of the biggest differences I noticed when I first started reading FreeRTOS was the adherence to a single exit point for each function.  I came to learn that this and many other unfamiliar practices stem from MISRA compliance (i.e., Rule 15.5).

Since it's been a week, I'm going to close this.  If anyone has a specific recommendation and rationale for a change, please feel free to open a new issue.

closed 2021-06-24 by dan4thewin