FreeRTOS/FreeRTOS #650: [BUG] the code can not build with C++ complier, reboot error

orkfire, 2021-07-16:
**Describe the bug**
My project is usually written with c++ code, so I want to compiler the freertos with c++ compiler. My develop tool is IAR FOR ARM9.10.1, and I download the latest FreeRTOS code, and open the demo<FreeRTOSv202104.00\FreeRTOS\Demo\CORTEX_M4F_STM32F407ZG-SK>, it is build well with C compiler, but when I use C++ complier, it report errors, 
**Target**
- Development board: [e.g. HiFive11 RevB]
- Instruction Set Architecture: [e.g. RV32IMAC]
- IDE and version: [e.g. Freedom Studio 4.12.0.2019-08-2]
- Toolchain and version: [e.g. riscv64-unknown-elf-gcc-8.3.0-2019.08.0]

**Host**
- Host OS: [e.g. MacOS]
- Version: [e.g. Mojave 10.14.6]

**To Reproduce**
- Use project ... and configure with ...
- Run on ... and could observe ... 

**Expected behavior**
A concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
e.g. code snippet to reproduce the issue.
e.g. stack trace, memory dump, debugger log, and many etc.

muneebahmed10, 2021-07-16:
Hi @orkfire,

If you are using C code, then unless it's already compatible with C++, it will need to be compiled as C. Your compiler is likely treating the C input as C++, so you may have to specify the language for your input files. For example, simple programs compiled with `g++` can have language specified with `-x`:
```bash
g++ *.cpp -x c *.c
```
Or, you could compile your C and C++ code separately and then use a C++ linker, e.g.
```bash
gcc -c *.c
g++ -c *.cpp
g++ *.o
```

Additionally, the C code will need `extern "C" {}` guards to avoid the C++ compiler mangling names when calling C functions from C++. The FreeRTOS kernel header files already include this, but some library files may not, so you should use the format
```c++
extern "C" {
    #include "some_c_header.h"
}
```
when including C header files that don't already have those guards.

orkfire, 2021-07-17:
> Hi @orkfire,
> 
> If you are using C code, then unless it's already compatible with C++, it will need to be compiled as C. Your compiler is likely treating the C input as C++, so you may have to specify the language for your input files. For example, simple programs compiled with `g++` can have language specified with `-x`:
> 
> ```shell
> g++ *.cpp -x c *.c
> ```
> 
> Or, you could compile your C and C++ code separately and then use a C++ linker, e.g.
> 
> ```shell
> gcc -c *.c
> g++ -c *.cpp
> g++ *.o
> ```
> 
> Additionally, the C code will need `extern "C" {}` guards to avoid the C++ compiler mangling names when calling C functions from C++. The FreeRTOS kernel header files already include this, but some library files may not, so you should use the format
> 
> ```c++
> extern "C" {
>     #include "some_c_header.h"
> }
> ```
> 
> when including C header files that don't already have those guards.

I tried using the extern "c" method, but it doesn't help, now I have a method to deal with it, First I using IAR build a static library using the c compiler, and then link it using c++ complier, it can work. but I don't think it is a good method when building a large project.

muneebahmed10, 2021-07-18:
Hi @orkfire,

Adding `extern "C"` doesn't automatically tell a compiler to compile specific code as C, it only tells a C++ compiler to not mangle certain C functions in C++ code so that compiled C code can be properly linked. You will still have to tell your compiler to compile C code with a C compiler and C++ with a C++ compiler. In the screenshot you posted, have you tried using the language setting "Auto (extension-based)"?

Additionally, have you taken a look at [IAR's notes on mixing C and C++](https://www.iar.com/knowledge/support/technical-notes/compiler/mixing-c-and-c/)? One recommendation there is
> If you want to use C code from your C++ application, consider building the C code in a separate (library) project. You can place the C (library) project in the same workspace as your C++ project for fast access.

orkfire, 2021-07-18:
> Hi @orkfire,
>
> Adding `extern "C"` doesn't automatically tell a compiler to compile specific code as C, it only tells a C++ compiler to not mangle certain C functions in C++ code so that compiled C [...]

I tried the auto option, it doesn't work,  if I using cmake or makefile, it may be very easy to organize the c and c++ file with a different compiler, but the IDE(IAR) is not that flexible. Maybe I should try using none IDE method to organize my project, os it will be very convenient.

muneebahmed10, 2021-07-21:
Hi @orkfire,

Do you have any further questions on this, or can we close this issue?

Thanks

closed 2021-07-21 by orkfire