FreeRTOS/FreeRTOS #792: [question] `mtvec` is not set in the demonstration of Spike?

kaizsv, 2022-02-15:
https://github.com/FreeRTOS/FreeRTOS/blob/449c7b29ca2b69b060ea11abc90fc94c8387318c/FreeRTOS/Demo/RISC-V-spike-htif_GCC/main.c#L43-L54

I think this demonstration doesn't set `mtvec` since I don't see any context switching with my GDB on Spike.
I can see the output correctly after adding either of the modifications to the `main(void)` function.

`__asm__ volatile( "csrw mtvec, %0" :: "r"( freertos_risc_v_trap_handler ) );`

or

(Compile with [vector_table.S](https://github.com/FreeRTOS/FreeRTOS/blob/main/FreeRTOS/Demo/RISC-V_RV32_SiFive_HiFive1-RevB_FreedomStudio/vector_table.S))
`__asm__ volatile( "csrw mtvec, %0" :: "r"( ( uintptr_t )freertos_vector_table | 0x1 ) );`

aggarg, 2022-02-16:
Yes, that change seem correct for the new re-factored code. Would you please raise a PR to add the following line:

```
__asm__ volatile( "csrw mtvec, %0" :: "r"( freertos_risc_v_trap_handler ) );
```

aggarg, 2022-02-17:
Thank you for the PR.

closed 2022-02-17 by aggarg