FreeRTOS/FreeRTOS #810: [Feature Request] xStreamBufferDiscard() macro to remove X bytes from a buffer

moefear85, 2022-04-25:
**Is your feature request related to a problem? Please describe.**
Sometimes I use buffers/queues as backups for data that might be re-/needed just-in-case. When it is clear it is no longer needed, I want to discard X many bytes/elements from the buffer/queue.

**Describe the solution you'd like**
A simple pop/discard function that simply advances the internal pointers of buffers/queues, defacto discarding what was before.

**Describe alternatives you've considered**
Currently I have to manually create a temporary array on the heap, copy into it, then delete, which is inefficient.

**Additional context**
Equally beneficial would be able to read any arbitrary segment of existing data/items in the buffer/queue, while keeping the data in the buffer. Although queues have a peek() functionality, it is limited, and it is totally missing for other primitives such as stream buffers.

paulbartell, 2022-04-25:
@moefear85 Thanks for the feature request. I can't commit to a particular timeline for this feature, but I will bring it up with the rest of the dev team.

moefear85, 2022-04-27:
thx.... I've implemented my own mods to the esp-idf port of freertos. I can provide them as patches if relevant. I can't PR as I'm still new to git.