FreeRTOS/FreeRTOS #815: [Feature Request] Is there any plans to port XT804

wyl2008056, 2022-05-17:
XT804 is a risk-v cpu arch，used in W801， w806, w800 ...

n9wxu, 2022-05-20:
There is a lot of activity on RISC-V happening.  Currently there is not a port in the official kernel repo but there are ports of FreeRTOS on the XT804 outside of the official FreeRTOS repo's.
One example is here: [https://github.com/IOsetting/wm-sdk-w806](https://github.com/IOsetting/wm-sdk-w806)
We always welcome contributions from the community.

wyl2008056, 2022-05-20:
> There is a lot of activity on RISC-V happening. Currently there is not a port in the official kernel repo but there are ports of FreeRTOS on the XT804 outside of the official FreeRTOS repo's. One [...]

hi, n9wxu, thank you for your reply.

closed 2022-05-20 by wyl2008056