FreeRTOS/FreeRTOS #824: how to porting the guix to freertos?

Barfell, 2022-06-22:
there has a question, some guix wiget need dynamic memeory, need the tx_byte_pool_create() function, but the function in the threadx.
how to fix it?

/* create a byte pool to use for rotating the needle bitmap */
tx_byte_pool_create(&memory_pool, "guix", guix_pool_memory, 1024);

/* install our memory allocation / de-allocation routines */
gx_system_memory_allocator_set(memory_allocate, memory_free);
I'm defined a text prompt,and a timer,in the timer int, refresh the text. the text not change.
but ,test guix + theadx,the text refreshed. because used the function tx_byte_pool_create();

UINT _cbEventWindow_language(GX_WINDOW *widget, GX_EVENT *event_ptr)
{
static uint32_t count = 6;
char buf[1] = {0};

switch (event_ptr->gx_event_type)
{
case GX_EVENT_SHOW:

    gx_system_timer_start((GX_WIDGET *)widget, GUI_ID_Timer0, 1, GX_TICKS_SECOND);

    gx_window_event_process(widget, event_ptr);
    break;
case GX_EVENT_TIMER:
    if (event_ptr->gx_event_payload.gx_event_timer_id == GUI_ID_Timer0)
    {
        sprintf(buf, "%d", count--);
        gx_prompt_text_set((GX_PROMPT *)&(window_language.window_language_prompt_countdown), buf);
    }
    break;

default:
    return gx_window_event_process(widget, event_ptr);
}

return 0;
}

paulbartell, 2022-06-22:
@Barfell : I'm not aware of any existing guix port for FreeRTOS.

Recent versions of FreeRTOS do not have built-in support for memory pools ( like the one allocated by a call to tx_byte_pool_create in your example code).

You could simply pass pvPortMalloc and vPortFree when calling gx_system_memory_allocator_set. Depending on your application, heap fragmentation could be a problem. In that case, you may very well want to implement a memory pool of some kind.

The [CMSIS-FreeRTOS package](https://github.com/ARM-software/CMSIS-FreeRTOS) from ARM includes a [memory pool implementation](https://github.com/ARM-software/CMSIS-FreeRTOS/blob/develop/CMSIS/RTOS2/FreeRTOS/Source/cmsis_os2.c#L2339) which you might find useful.

Barfell, 2022-06-23:
Mabybe use the Family Big Box: threadx + guix. the memory management service may seem like a bit of a hassle。
thank a lot . I`m try to find the  CMSIS-FreeRTOS package.

paulbartell, 2022-06-23:
Closing this issue out. Feel free to re-open it if you have any related followup questions.

closed 2022-06-23 by paulbartell

Barfell, 2022-06-25:
in the threadx:
tx_byte_allocate(&guix_pool, &memptr, size, TX_NO_WAIT)

in the freertos v2:
memptr =osMemoryPoolAlloc(guix_pool, 0U);

the threadx specify size, but the freertos v2 not.