FreeRTOS/FreeRTOS #827: Whether it is a deviation of rule 20.2 of misra C

Joshua1023, 2022-07-04:
In `FreeRTOS/Source/include/deprecated_definitions.h`, there are several places like:

line 43:
```c
#include "..\..\Source\portable\owatcom\16bitdos\pc\portmacro.h"
```

However, according to rule 20.2 of misra C, character ` \`  shall not occur in a header file name. Is there an intended deviation in such cases?

Skptak, 2022-07-05:
Hey,

Thanks for reaching out, this would be an intended deviation in this case. This deviation is needed for compilation on windows based systems.

Joshua1023, 2022-07-25:
Thx for your answer. But I still have a question: even on Windows, `/` in the #include directive can still work. Is it necessary to use '\' in these #include directives?

johnrhen, 2022-07-25:
Determining this with confidence would require checking each compiler, but it is likely true that replacing `\` with `/` would function on the platforms we support. However, we do not intend to modify the deprecated file to remove the minor deviation.

closed 2022-07-26 by Joshua1023