FreeRTOS/FreeRTOS #830: target emulation 'elf32-littleriscv' does not match 'elf64-littleriscv‘

lookcat, 2022-07-18:
**Describe the bug**
As far as I know, this is due to the linker expecting natively ELF64 objects, whereas everything was built rv32imac/ilp32, this producing ELF32 binaries.

**Target**
- Development board: qemu
- Instruction Set Architecture: RV32IMAC
- IDE and version: vscode
- Toolchain and version: riscv64-unknown-elf-gcc-8.3.0-2019.08.0

**Host**
- Host OS: [e.g. MacOS]
- Version: [e.g. Mojave 10.14.6]

**Project**
riscv-v-qemu-virt-gcc

**bugfix**
ASFLAGS = -march=rv32imac -mabi=ilp32 -mcmodel=medany
ARCH_FLAGS = -march=rv32imac -mabi=ilp32 -mcmodel=medany
LDFLAGS = -nostartfiles -Tfake_rom.lds \
	-Xlinker --gc-sections \
	-Xlinker --defsym=__stack_size=300 \
	${ASFLAGS}
[https://github.com/stanley7342/riscv_freertos/blob/master/BuildEnvironment.mk](url)

alfred2g, 2022-07-21:
Hello,
what exactly are you trying to do? where is the bug? how to reproduce?
the link provided is broken, had to copy the text to get there.

lookcat, 2022-07-21:
(base) tom@192 RISC-V-Qemu-virt_GCC % make
riscv64-unknown-elf-gcc -D__riscv_float_abi_soft -DportasmHANDLE_INTERRUPT=handle_trap -I . -I ../Common/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions -march=rv32imac -mabi=ilp32 -mcmodel=medany -Wall -fmessage-length=0 -ffunction-sections -fdata-sections -fno-builtin-printf -O2 -MMD -MP -c main.c -o build/main.o
riscv64-unknown-elf-gcc -D__riscv_float_abi_soft -DportasmHANDLE_INTERRUPT=handle_trap -I . -I ../Common/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions -march=rv32imac -mabi=ilp32 -mcmodel=medany -Wall -fmessage-length=0 -ffunction-sections -fdata-sections -fno-builtin-printf -O2 -MMD -MP -c main_blinky.c -o build/main_blinky.o
riscv64-unknown-elf-gcc -D__riscv_float_abi_soft -DportasmHANDLE_INTERRUPT=handle_trap -I . -I ../Common/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions -march=rv32imac -mabi=ilp32 -mcmodel=medany -Wall -fmessage-length=0 -ffunction-sections -fdata-sections -fno-builtin-printf -O2 -MMD -MP -c riscv-virt.c -o build/riscv-virt.o
riscv64-unknown-elf-gcc -D__riscv_float_abi_soft -DportasmHANDLE_INTERRUPT=handle_trap -I . -I ../Common/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions -march=rv32imac -mabi=ilp32 -mcmodel=medany -Wall -fmessage-length=0 -ffunction-sections -fdata-sections -fno-builtin-printf -O2 -MMD -MP -c ns16550.c -o build/ns16550.o
riscv64-unknown-elf-gcc -D__riscv_float_abi_soft -DportasmHANDLE_INTERRUPT=handle_trap -I . -I ../Common/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions -march=rv32imac -mabi=ilp32 -mcmodel=medany -Wall -fmessage-length=0 -ffunction-sections -fdata-sections -fno-builtin-printf -O2 -MMD -MP -c /Users/tom/Documents/FreeRTOS/FreeRTOS/Demo/Common/Minimal/EventGroupsDemo.c -o build//Users/tom/Documents/FreeRTOS/FreeRTOS/Demo/Common/Minimal/EventGroupsDemo.o
riscv64-unknown-elf-gcc -D__riscv_float_abi_soft -DportasmHANDLE_INTERRUPT=handle_trap -I . -I ../Common/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions -march=rv32imac -mabi=ilp32 -mcmodel=medany -Wall -fmessage-length=0 -ffunction-sections -fdata-sections -fno-builtin-printf -O2 -MMD -MP -c /Users/tom/Documents/FreeRTOS/FreeRTOS/Demo/Common/Minimal/TaskNotify.c -o build//Users/tom/Documents/FreeRTOS/FreeRTOS/Demo/Common/Minimal/TaskNotify.o
riscv64-unknown-elf-gcc -D__riscv_float_abi_soft -DportasmHANDLE_INTERRUPT=handle_trap -I . -I ../Common/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions -march=rv32imac -mabi=ilp32 -mcmodel=medany -Wall -fmessage-length=0 -ffunction-sections -fdata-sections -fno-builtin-printf -O2 -MMD -MP -c /Users/tom/Documents/FreeRTOS/FreeRTOS/Demo/Common/Minimal/TimerDemo.c -o build//Users/tom/Documents/FreeRTOS/FreeRTOS/Demo/Common/Minimal/TimerDemo.o
riscv64-unknown-elf-gcc -D__riscv_float_abi_soft -DportasmHANDLE_INTERRUPT=handle_trap -I . -I ../Common/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions -march=rv32imac -mabi=ilp32 -mcmodel=medany -Wall -fmessage-length=0 -ffunction-sections -fdata-sections -fno-builtin-printf -O2 -MMD -MP -c /Users/tom/Documents/FreeRTOS/FreeRTOS/Demo/Common/Minimal/blocktim.c -o build//Users/tom/Documents/FreeRTOS/FreeRTOS/Demo/Common/Minimal/blocktim.o
riscv64-unknown-elf-gcc -D__riscv_float_abi_soft -DportasmHANDLE_INTERRUPT=handle_trap -I . -I ../Common/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/include -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V -I /Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V/chip_specific_extensions/RV32I_CLINT_no_extensions -march=rv32imac -mabi=ilp32 -mcmodel=medany -Wall -fmessage-length=0 -ffunction-sections -fdata-sections -fno-builtin-printf -O2 -MMD -MP -c /Users/tom/Documents/FreeRTOS/FreeRTOS/Demo/Common/Minimal/dynamic.c -o build//Users/tom/Documents/FreeRTOS/FreeRTOS/Demo/Common/Minimal/dynamic.o
[... cut ...]
  target emulation `elf32-littleriscv' does not match `elf64-littleriscv'
/Users/tom/embed/riscv-tool/riscv64-unknown-elf-gcc-8.3.0/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file build//Users/tom/Documents/FreeRTOS/FreeRTOS/Source/portable/GCC/RISC-V/portASM.o
make: *** [build/RTOSDemo.axf] Error 1
(base) tom@192 RISC-V-Qemu-virt_GCC % 

> Hello, what exactly are you trying to do? where is the bug? how to reproduce? the link provided is broken, had to copy the text to get there.

alfred2g, 2022-07-26:
I remember seeing a similar issue (where the linker crashes) when i modify the linker script to some bad values.. did you modify the linker script ?

alfred2g, 2022-08-08:
Closing the issue due to no activity, please reopen if there are any new information.

closed 2022-08-08 by alfred2g

TommyMurphyTM1234, 2022-09-26:
I strongly suspect that the issue here is that the RISC-V compiler toolchain does not provide a multilib for rv32imac, so when the rv32 object files are being linked with system libraries the toolchain falls back to the toolchain default architecture (usually rv64gc which is the default toolchain architecture unless overridden at configure time) and that's where the mismatch between rv32 object files and rv64 libraries arises. If that's the issue then either the toolchain was not configured with `--enable-multilib` or the default set of multilibs (`gcc/gcc/config/riscv/t-elf-multlib`) does not specify rv32imac.

ezhes, 2022-10-07:
Ah! Finally figured this out (for my setup, anyways). I have a multilib gcc toolchain and, as Tommy suggested, the issue is that the linker is defaulting to rv64gc which is not what we want. To fix this, I modified the Makefile's `LDFLAGS` to include `-march=rv32imac -mabi=ilp32`, which instructs it to link as rv32 as desired.

TommyMurphyTM1234, 2022-10-07:
Ah, ok, I was close but not correct. I overlooked the fact that the linker step was not being passed `-march= rv32imac -mabi=ilp32` and, so, was implicitly falling back to the toolchain defaults `-march=rv64gc -mabi=lp64d`. And not that the toolchain was missing the `-march= rv32imac -mabi=ilp32` multilib.
> `riscv64-unknown-elf-gcc -nostartfiles -Tfake_rom.lds -Xlinker --gc-sections -Xlinker --defsym=__stack_size=300  build/main.o ...`

eminedrmsky, 2023-10-21:
Hi, I am getting a similar error which says "target emulation `elf64-littleriscv' does not match `elf32-littleriscv' ", I am trying to build Pk and I am using a riscv64-unknown-elf toolchain which is multilib. When I try to use toolchain with my 32-bit core which I wrote with Verilog, it works fine. But when I try to build Pk, it throws that error, I am using  "./configure --prefix=$RISCV --host=riscv64-unknown-elf --with-arch=rv32gc" for configuration of Pk.

TommyMurphyTM1234, 2023-10-21:
> Hi, I am getting a similar error which says "target emulation `elf64-littleriscv' does not match `elf32-littleriscv' ", I am trying to build Pk and I am using a riscv64-unknown-elf toolchain which [...]

Where did you get the toolchain?
What does the following report?
```
riscv64-unknown-elf-gcc -print-multi-lib
```
I suspect that the `rv32gc/ilp32d` multilib will probably actually be `rv32gc_zicsr/ilp32d` or `rv32imafdc_zicsr/ilp32d`.

See here:

* https://github.com/riscv-collab/riscv-gnu-toolchain/issues/1315

eminedrmsky, 2023-10-21:
I am using the official riscv toolchain : https://github.com/riscv-collab/riscv-gnu-toolchain
I configured it for rv32i and rv32im. I enabled multilib.
Right now, I am trying to build Pk, I am configuring Pk with "--with-arch=rv32i;rv32im", and I am getting error "Error: unrecognized opcode `csrr a0,cycle', extension `zicsr' required". I solved this one by using "--with-arch=rv32i_zicsr_zifencei;rv32im_zicsr_zifencei". Then I got both "ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'" and "undefined reference to `__mulsi3'
collect2: error: ld returned 1 exit status". I tried to solve the second error by deleting -nostdlib from makefile, but I kept getting same error. Here is the full report:

./config.status
config.status: creating pk.mk
config.status: creating bbl.mk
config.status: creating softfloat.mk
config.status: creating dummy_payload.mk
config.status: creating machine.mk
config.status: creating util.mk
config.status: creating Makefile
config.status: creating config.h
config.status: config.h is unchanged
riscv64-unknown-elf-gcc -Wl,--build-id=none -static -nostartfiles  -march=rv32i_zicsr_zifencei -mabi=ilp32 -o pk pk.o -L.  -lpk  -lmachine  -lsoftfloat  -lutil -lgcc -T ../pk/pk.lds
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(muldi3.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(muldi3.o)
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(div.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf64-littleriscv' does not match `elf32-littleriscv'
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(div.o)
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: ./libmachine.a(muldiv_emulation.o): in function `.L13':
muldiv_emulation.c:(.text+0xf8): undefined reference to `__mulsi3'
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: ./libsoftfloat.a(f32_sqrt.o): in function `.L11':
f32_sqrt.c:(.text+0xf8): undefined reference to `__mulsi3'
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: ./libsoftfloat.a(s_approxRecip32_1.o): in function `.L0 ':
s_approxRecip32_1.c:(.text+0x44): undefined reference to `__mulsi3'
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: ./libsoftfloat.a(s_approxRecipSqrt32_1.o): in function `.L0 ':
s_approxRecipSqrt32_1.c:(.text+0x54): undefined reference to `__mulsi3'
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: s_approxRecipSqrt32_1.c:(.text+0x78): undefined reference to `__mulsi3'
collect2: error: ld returned 1 exit status
make: *** [Makefile:319: pk] Error 1

I am a fresher with GCC and Riscv.

eminedrmsky, 2023-10-21:
I also got the spike and Pk from https://github.com/riscv-software-src/riscv-tools

TommyMurphyTM1234, 2023-10-21:
You didn't answer my question about `-print-multi-lib`.
The issue is that the multilib that you're requesting isn't being matched/found so the toolchain falls back to linking the default libs for `rv64gc/lp64d` hence the link errors.

You need to build a toolchain for the specific `arch/abi` that you need or a multilib toolchain that supports the specific `arch/abi` that you need.

But I'm confused as to what `arch/abi`(s?) that you're trying to use as you mention a whole bunch of them.

The splitting of the RISC-V `Zicsr` and `Zifencei` extensions from the base `I` integer ISA caused a lot of confusion with libraries/multilibs as evidenced by the thread that I linked to above.

TommyMurphyTM1234, 2023-10-21:
> `riscv64-unknown-elf-gcc -Wl,--build-id=none -static -nostartfiles -march=rv32i_zicsr_zifencei -mabi=ilp32 -o pk pk.o -L. -lpk -lmachine -lsoftfloat -lutil -lgcc -T ../pk/pk.lds
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/libgcc.a(muldi3.o): ABI is incompatible with that of the selected emulation:
target emulation elf64-littleriscv' does not match elf32-littleriscv'`

Almost certainly this means that your toolchain doesn't include multilibs for `rv32i_zicsr_zifencei/ilp32`. If you need that specific multilib then you need to build the toolchain as follows:
```
# build toolchain specifically for rv32i_zicsr_zifencei/ilp32
./configure --with-arch=rv32i_zicsr_zifencei --with-abi=ilp32 ...

Or

# build toolchain with multilib for rv32i_zicsr_zifencei/ilp32
./configure ... --with-multilib-generator="rv32i_zicsr_zifencei-ilp32--"

eminedrmsky, 2023-10-22:
I rebuild the toolchain as you suggested, and it works now. Thank you!

TommyMurphyTM1234, 2023-10-22:
> I rebuild the toolchain as you suggested, and it works now. Thank you!

Glad it helped and that you made progress.

The RISC-V stuff can be confusing - including the `Zicsr/Zifencei` extensions being split out from the original base `I` integer ISA as separate extensions - even though `Zicsr` is almost always needed in practice anyway (at least for embedded) - and the more recent changes (in GCC 13 I think) for "fuzzy" matching a specified `-march/-mabi` tuple with an available compatible multilib even if there is no exact match.