FreeRTOS/FreeRTOS #834: [Feature Request] Can SMP RP2040 demo core0 and core1 both run freertos?

thomas-coding, 2022-08-02:
Hi,

I want to know whether multiple cores can run FreeRTOS at the same time. If yes, how to modify RP2040 demo to support this. Thanks.

From SMP demo RP2040, only one core run FreeRTOS. 

FreeRTOS-SMP-Demos/FreeRTOS/Demo/CORTEX_M0+_RP2040/README.md
```
### Multicore Demo
The multicore demo application runs FreeRTOS tasks on one core which interacts
with the code running on the other core using Raspberry Pico SDK synchronization
primitives. There are two versions of the same demo - One version runs FreeRTOS
on core 0, the other runs FreeRTOS on core 1.
```

dachalco, 2022-08-03:
Hi @thomas-coding 

As in each core runs a single-core FreeRTOS, or rather that both core participate in running the scheduler?

thomas-coding, 2022-08-04:
Hi dachalco,

I don't want to run pico SDK,  I want core0 and core1 both running FreeRTOS task at the same time, but not just signle-core FreeRTOS, I think its called "SMP", only one scheduler, and can assign task to core0 or core1 according to some rules. like there are three task (task1, taks2, task3). All these task is ready for run. After scheduler start, scheduler assign task1 to core0, and core0 running task1. scheduler assign task2 to core1, and core1 running task2. Then after a while, task1 enter the block state. scheduler can assign task3 to core1.

Can now support this? If yes, how to modify RP2040 demo to support this.

Thanks.

thomas-coding, 2022-08-06:
Hi,

I studied the code and made some attempts. It's OK now.

It need create new demo, copy OnEitherCore to OnDualCore, and mainly change is:

FreeRTOS/FreeRTOS/Demo/CORTEX_M0+_RP2040/OnDualCore/FreeRTOSConfig.h
```
#define configNUM_CORES                         2
```
Define two core both running FreeRTOS

FreeRTOS/FreeRTOS/Demo/CORTEX_M0+_RP2040/OnDualCore/main.c
```
//    multicore_launch_core1(prvCore1Entry);
//#if ( mainRUN_FREE_RTOS_ON_CORE == 0 )
    prvLaunchRTOS();
//#else
//    prvNonRTOSWorker();
//#endif
```
Not start core1 here because running on both core case, when core0 start scheduler, it will launch core1 .
Core1 running RTOS.

thomas-coding, 2022-08-06:
Close

closed 2022-08-06 by thomas-coding