FreeRTOS/FreeRTOS #855: [DOC] Demo 'on-windows-using-msys2' miss miDebuggerPath setting

tiantian1645, 2022-10-10:
**Describe the issue**
When debug [on-windows-using-msys2](https://github.com/FreeRTOS/FreeRTOS/tree/main/FreeRTOS/Demo/Posix_GCC#on-windows-using-msys2)  lost gdb path setting. And before debug shold install `make` and `gdb` , so I use `pacman -S gcc make gdb`

**Reference**
https://github.com/FreeRTOS/FreeRTOS/tree/main/FreeRTOS/Demo/Posix_GCC#on-windows-using-msys2

**Browser**
- Browser: [Chrome]
- Version: [106.0.5249.103]

archigup, 2022-10-13:
Hey thanks for reaching out, would you be able additional information?
To confirm, you are trying to run the Posix_GCC demo on Windows using msys2?
Are you having trouble running the demo, or is the demo running and using gdb is the issue?
Could you post the output you are getting?

n9wxu, 2022-10-27:
Closing this issue after PR #859

closed 2022-10-27 by n9wxu