FreeRTOS/FreeRTOS #858: [BUG] Compilation fails with -Wall -Werror -Wextra -O3 when using arm gcc 11.1 and up

RyanHanson-Deere, 2022-10-13:
**Describe the bug**
tasks.c is failing to build when using arm gcc 11.2 (The compiler distributed with Xilinx) and the options -Wall -Werror -Wextra -O3

Here is the error I get:
```
In file included from /workspace/VanillaR5Alt/R5Common/bsp/r5_0.bsp/psu_cortexr5_0/libsrc/freertos10_xilinx_v1_11/src/task.h:37,                                                                            
                 from /workspace/VanillaR5Alt/R5Common/bsp/r5_0.bsp/psu_cortexr5_0/libsrc/freertos10_xilinx_v1_11/src/tasks.c:40:                                                                           
/workspace/VanillaR5Alt/R5Common/bsp/r5_0.bsp/psu_cortexr5_0/libsrc/freertos10_xilinx_v1_11/src/tasks.c: In function 'xTaskGetHandle':                                                                      
/workspace/VanillaR5Alt/R5Common/bsp/r5_0.bsp/psu_cortexr5_0/libsrc/freertos10_xilinx_v1_11/src/list.h:289:64: error: array subscript 'ListItem_t {aka struct xLIST_ITEM}[0]' is partly outside array bounds
 of 'List_t[1]' {aka 'struct xLIST[1]'} [-Werror=array-bounds]                                                                                                                                              
  289 |             ( pxConstList )->pxIndex = ( pxConstList )->pxIndex->pxNext;                       \                                                                                                    
      |                                        ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~                                                                                                                             
/workspace/VanillaR5Alt/R5Common/bsp/r5_0.bsp/psu_cortexr5_0/libsrc/freertos10_xilinx_v1_11/src/tasks.c:2412:13: note: in expansion of macro 'listGET_OWNER_OF_NEXT_ENTRY'                                  
 2412 |             listGET_OWNER_OF_NEXT_ENTRY( pxFirstTCB, pxList ); /*lint !e9079 void * is used as this macro is used with timers and co-routines too.  Alignment is known to be fine as the type of the
 pointer stored and retrieved is the same. */                                                                                                                                                               
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                             
/workspace/VanillaR5Alt/R5Common/bsp/r5_0.bsp/psu_cortexr5_0/libsrc/freertos10_xilinx_v1_11/src/tasks.c:362:35: note: while referencing 'xSuspendedTaskList'                                                
  362 |     PRIVILEGED_DATA static List_t xSuspendedTaskList; /*< Tasks that are currently suspended. */                                                                                                    
      |                                   ^~~~~~~~~~~~~~~~~~                                                                                                                                                
In file included from /workspace/VanillaR5Alt/R5Common/bsp/r5_0.bsp/psu_cortexr5_0/libsrc/freertos10_xilinx_v1_11/src/task.h:37,                                                                            
                 from /workspace/VanillaR5Alt/R5Common/bsp/r5_0.bsp/psu_cortexr5_0/libsrc/freertos10_xilinx_v1_11/src/tasks.c:40:                                                                           
/workspace/VanillaR5Alt/R5Common/bsp/r5_0.bsp/psu_cortexr5_0/libsrc/freertos10_xilinx_v1_11/src/list.h:289:64: error: array subscript 'ListItem_t {aka struct xLIST_ITEM}[0]' is partly outside array bounds of 'List_t[1]' {aka 'struct xLIST[1]'} [-Werror=array-bounds]                                                                                                                                              
  289 |             ( pxConstList )->pxIndex = ( pxConstList )->pxIndex->pxNext;                       \                                                                                                    
      |                                        ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~                                                                                                                            
/workspace/VanillaR5Alt/R5Common/bsp/r5_0.bsp/psu_cortexr5_0/libsrc/freertos10_xilinx_v1_11/src/tasks.c:2416:17: note: in expansion of macro 'listGET_OWNER_OF_NEXT_ENTRY'                                  
 2416 |                 listGET_OWNER_OF_NEXT_ENTRY( pxNextTCB, pxList ); /*lint !e9079 void * is used as this macro is used with timers and co-routines too.  Alignment is known to be fine as the type of 
the pointer stored and retrieved is the same. */                                                                                                                                                            
      |                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                         
/workspace/VanillaR5Alt/R5Common/bsp/r5_0.bsp/psu_cortexr5_0/libsrc/freertos10_xilinx_v1_11/src/tasks.c:362:35: note: while referencing 'xSuspendedTaskList'                                                
  362 |     PRIVILEGED_DATA static List_t xSuspendedTaskList; /*< Tasks that are currently suspended. */                                                                                                    
      |                                   ^~~~~~~~~~~~~~~~~~                                                                                                                                                
In file included from /workspace/VanillaR5Alt/R5Common/bsp/r5_0.bsp/psu_cortexr5_0/libsrc/freertos10_xilinx_v1_11/src/task.h:37,                                                                            
                 from /workspace/VanillaR5Alt/R5Common/bsp/r5_0.bsp/psu_cortexr5_0/libsrc/freertos10_xilinx_v1_11/src/tasks.c:40:                                                                           
/workspace/VanillaR5Alt/R5Common/bsp/r5_0.bsp/psu_cortexr5_0/libsrc/freertos10_xilinx_v1_11/src/list.h:289:64: error: array subscript 'ListItem_t {aka struct xLIST_ITEM}[0]' is partly outside array bound$ of 'List_t[1]' {aka 'struct xLIST[1]'} [-Werror=array-bounds]                                                                                                                                              
  289 |             ( pxConstList )->pxIndex = ( pxConstList )->pxIndex->pxNext;                       \                                                                                                    
      |                                        ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~                                                                                                                             
/workspace/VanillaR5Alt/R5Common/bsp/r5_0.bsp/psu_cortexr5_0/libsrc/freertos10_xilinx_v1_11/src/tasks.c:2412:13: note: in expansion of macro 'listGET_OWNER_OF_NEXT_ENTRY'                                  
 2412 |             listGET_OWNER_OF_NEXT_ENTRY( pxFirstTCB, pxList ); /*lint !e9079 void * is used as this macro is used with timers and co-routines too.  Alignment is known to be fine as the type of th$ pointer stored and retrieved is the same. */                                                                                                                                                               
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~                                                                                                                                                             
/workspace/VanillaR5Alt/R5Common/bsp/r5_0.bsp/psu_cortexr5_0/libsrc/freertos10_xilinx_v1_11/src/tasks.c:355:35: note: while referencing 'xTasksWaitingTermination'                                          
  355 |     PRIVILEGED_DATA static List_t xTasksWaitingTermination; /*< Tasks that have been deleted - but their memory not yet freed. */                                                                   
      |                                   ^~~~~~~~~~~~~~~~~~~~~~~~                                                                 
```

**Target**
- Development board: Xilinx Ultrascale
- Instruction Set Architecture: ARM
- IDE and version: VIM/CMake
- Toolchain and version: armr5-none-eabi-gcc

**Host**
- Host OS: Linux 
[... cut ...]
- Build tasks.c with gcc 11.2 with the  -Wall -Werror -Wextra -O3 flags
Or
- Godbolt link: https://godbolt.org/z/4q7cTMqzd
**Expected behavior**

tasks.c compiles

**Additional context**

I am mostly sure this is a compiler bug. But just in case it isn't I wanted to post it here.

The only other concern that I have is that `listGET_OWNER_OF_NEXT_ENTRY` is accessing the element after `xListEnd` and maybe that is bad. I don't know if `xListEnd->pxNext` gets set to the first element of the list or is kept pointing back to `xListEnd`.

archigup, 2022-10-13:
Hi, thanks for reaching out; I'll take a look at this

archigup, 2022-10-19:
I've investigated it further, and it does not seem like a compiler bug to me. We are investigating possible fixes. Thanks again for bringing this to our attention!

As for your concern about `listGET_OWNER_OF_NEXT_ENTRY`, that is fine. The list is a circular list, so the element after `xListEnd` is the first element of the list. Where this becomes an issue is that if the list is empty, the element after `xListEnd` is itself.

archigup, 2022-11-30:
https://gcc.gnu.org/onlinedocs/gcc/Warning-Options.html#index-Warray-bounds

Looking at above gcc docs explains why it only happens at those optimization levels; the warning depends on `-ftree-vrp` which is not on all optimization levels. I am able to reproduce the warning at `-00` when `-ftree-vrp` is enabled, and not able to reproduce with  `-03 -fno-tree-vrp`.

archigup, 2022-12-15:
Closing as fix was merged

closed 2022-12-15 by archigup