FreeRTOS/FreeRTOS #865: [BUG] Can't prove queue/create.c with VeriFast

master-q, 2022-10-26:
**Describe the bug**
Can't prove queue/create.c with VeriFast.

**Target**
- Development board: Any
- Instruction Set Architecture: Any
- IDE and version: VeriFast 19.12 for C and Java (released 2019-12-02)
- Toolchain and version: VeriFast 19.12 for C and Java (released 2019-12-02)

**Host**
- Host OS: Debian GNU/Linux amd64
- Version: unstable

**To Reproduce**
- git clone git@github.com:FreeRTOS/FreeRTOS.git
- cd FreeRTOS/FreeRTOS/Test/VeriFast
- make

**Expected behavior**
We should see following message:

```
0 errors found (??? statements verified)
```

**Screenshots**

```
$ verifast -version|&grep released
VeriFast 19.12 for C and Java (released 2019-12-02) <https://github.com/verifast/verifast/>
$ make
queue/create.c
include/proof/queue.h(187,2-3): Parse error.
Coverage regression failed: Expected 325 statements verified.
make: *** [Makefile:21: queue] Error 1
```

```
$ vfide -I include queue/create.c
```
[image]

**Additional context**
None.

master-q, 2022-10-26:
I believe this is caused by the miss style comments.
Attached patch works for queue/create.c.

[fix_miss_style_comments.patch.txt](https://github.com/FreeRTOS/FreeRTOS/files/9866407/fix_miss_style_comments.patch.txt)

May I create PR to fix this issue?

n9wxu, 2022-10-27:
We welcome any PR's. Please let me know when it is ready for review.

n9wxu, 2022-10-27:
This has been resolved with PR #836

closed 2022-10-27 by n9wxu