FreeRTOS/FreeRTOS #877: [BUG] PKCS#11 session close failure during disconnection

KazukiMochizuki926, 2022-11-21:
**Describe the bug**
It's about mbedtls_pk_pkcs11.c in the dev branch.
Since the following source code is cast with a structure suitable for ECDSA, it fails to close the PKCS#11 session when the key type is RSA.
We believe that you need to cast with a struct that matches the key you are using.
https://github.com/FreeRTOS/FreeRTOS/blob/dev/FreeRTOS-Plus/Source/Application-Protocols/network_transport/mbedtls_pk_pkcs11.c#L326
[image]

**Target**
- Development board: CK-RX65N
- Instruction Set Architecture: Renesas RX
- IDE and version: e2 studio-2022-10
- Toolchain and version: Renesas CC-RX v3.04.00

**Host**
- Host OS: Windows
- Version: Windows 10

**To Reproduce**
- We built a project for CK-RX65N based on corePKCS11_MQTT_Mutual_Auth_Windows_Simulator.
- An error occurs within the lPKCS11PkMbedtlsCloseSessionAndFree function for closing the connection.

**Expected behavior**
PKCS#11 session close fails when key type is RSA.

**Screenshots**
[image]

paulbartell, 2022-11-21:
@KazukiMochizuki926 Thanks for your report. I'll address this today.

KazukiMochizuki926, 2022-11-22:
Thanks for the bug fix.
However, `pxMbedtlsPkCtx->pk_info.type` and `pxMbedtlsPkCtx->pk_info.type` gave a compilation error.
I believe the code below is correct.

```c
        if( pxMbedtlsPkCtx->pk_info->type == MBEDTLS_PK_ECKEY )
        {
            pxP11Ctx = &( ( ( P11EcDsaCtx_t * ) ( pxMbedtlsPkCtx->pk_ctx ) )->xP11PkCtx );
        }
        else if( pxMbedtlsPkCtx->pk_info->type == MBEDTLS_PK_RSA )
        {
            pxP11Ctx = &( ( ( P11RsaCtx_t * ) ( pxMbedtlsPkCtx->pk_ctx ) )->xP11PkCtx );
        }
        else
        {
            pxP11Ctx = NULL;
            xResult = CKR_FUNCTION_FAILED;
        }
```

paulbartell, 2022-11-25:
@KazukiMochizuki926 Thanks much. We've integrated these changes into the dev branch.

PR #878 (2022-11-21): Fix bug in lPKCS11PkMbedtlsCloseSessionAndFree. Add doxygen api docs.
Description
-----------
Fix a bug in the lPKCS11PkMbedtlsCloseSessionAndFree function which did not work properly with RSA keys.

Add some doxygen api docs for public functions.

Fix #877 

By submitting this pull request, I confirm that you can use, modify, copy, and redistribute this contribution, under the terms of your choice.
closed 2022-11-25 by paulbartell