FreeRTOS/FreeRTOS #967: [BUG] Segment fault error caused by pxTopOfStack

xzwDavid, 2023-03-16:
Hi~
I use ubuntu 20.04 and run FreeRTOS_Plus_TCP_Echo_Posix on it.
However, it down because of segment error.

And I use gdb to debug it.

Found that this func " pxNewTCB->pxTopOfStack = pxPortInitialiseStack( pxTopOfStack, pxNewTCB->pxStack, pxTaskCode, pvParamet)" that produce some bug?

I used static way.

I also change some xTaskCreateStatic to xTaskCreate, however, it only works for some task while others don't.

adam-scislowicz, 2023-03-16:
Could you share a minimal code sample reproducing the issue?

adam-scislowicz, 2023-03-16:
Hi xzwDavid. We have applied a simple patch, increasing the stack size. Could you please verify that this addresses your issue?

xzwDavid, 2023-03-17:
Thanks for your responding!

[image]
It seems still wrong.Here is my debug info.
Sorry for I could not upload image last time.

xzwDavid, 2023-03-17:
Also, when I change static way to dynamic, it works for some at the beginning, however, it fails when called by Timer.c
[image]

xzwDavid, 2023-03-17:
Hi, I think I found the problem.
In ubuntu, the StackDepth would be set to zero.
And I don't know how to fix it.

PS:
I made some hard code for it, like set the onfigTIMER_TASK_STACK_DEPTH to big which is passed in "xTaskCreate". Such trick works.
[image]

xzwDavid, 2023-03-17:
After a lot of time and debugging, I think I found where the bug is.

In some linux-OS, the value of "PTHREAD_STACK_MIN" would be too big to fail the type casting.
Such problem could be fixed by change type casting to match bigger "PTHREAD_STACK_MIN".

I do want to commit a PR but It seems that I don't have permission. 
 : )

kar-rahul-aws, 2023-03-17:
Hi xzwDavid,
We have tried running the Demo on Ubuntu 20.04 LTS and 22.04 LTS , it runs properly without any segmentation error. Would you try re-running with a fresh clone and check if the issue still persists?
And when you say changing type casting to match bigger "PTHREAD_STACK_MIN", I assume you mean this line
https://github.com/FreeRTOS/FreeRTOS/blob/main/FreeRTOS-Plus/Demo/FreeRTOS_Plus_TCP_Echo_Posix/FreeRTOSConfig.h#L46

adam-scislowicz, 2023-03-17:
xzwDavid, could you post the FreeRTOSConfig.h you are using as well along with the toolchain you are using to compile if it is not the default one?

xzwDavid, 2023-03-17:
Sure!
[image]

[FreeRTOSConfig.zip](https://github.com/FreeRTOS/FreeRTOS/files/11003140/FreeRTOSConfig.zip)

adam-scislowicz, 2023-03-17:
It is possible (ASLR) address space layout randomization is causing the difference in results. That, a difference in the configs, or unknown. I will try to reproduce here.

adam-scislowicz, 2023-03-17:
I've tried to reproduce with your FreeRTOSConfig.h. I am running on a 64bit platform and so I have to add the following config otherwise an assertion was triggered, however it does keep executing with the assertion and it could mean under-allocated buffers which could result in a stack overflow:

`#define ipconfigBUFFER_PADDING (30)`

I also tried building adding -fstack-protector-string and running it through valgrind and wasn't able to find any other issue in my setup. Could you try the config above?

I'm testing with the same version of Ubuntu and gcc.

xzwDavid, 2023-03-18:
Hi~ Thanks for your guys time.

I just reclone it with command "git clone https://github.com/FreeRTOS/FreeRTOS.git --recurse-submodules" and navigate to Demo, make, cd build and run.

This bug still here and I still insist my opinion about it so I print the value.
[image]

There is a considerable difference that my ubuntu is arm architecture:
[image]

[Archive.zip](https://github.com/FreeRTOS/FreeRTOS/files/11007227/Archive.zip)

aggarg, 2023-03-18:
Your finding is correct. We should remove the `unsigned short` cast from here - https://github.com/FreeRTOS/FreeRTOS/blob/main/FreeRTOS-Plus/Demo/FreeRTOS_Plus_TCP_Echo_Posix/FreeRTOSConfig.h#L46

aggarg, 2023-03-18:
This PR address this - https://github.com/FreeRTOS/FreeRTOS/pull/971

Can you confirm that this fixes your issue?

aggarg, 2023-03-20:
I am closing this issue. Feel free to reopen or create a new one if you still face some issue.

closed 2023-03-20 by aggarg