FreeRTOS/FreeRTOS #972: [BUG] When vApplicationTickHook is in use, Cortex R5 UltraScale should save FPU context in tick handler

chris-laplante, 2023-03-20:
**Describe the bug**
We are running the [FreeRTOS demo](https://github.com/FreeRTOS/FreeRTOS/tree/main/FreeRTOS/Demo/CORTEX_R5_UltraScale_MPSoC) on our Xilinx ZCU104 dev board. Even after setting configUSE_TASK_FPU_SUPPORT = 2, the demo fails sporadically after a few seconds at the FPU register test (screenshot below).

The problem is that the demo's tick handler uses `sprintf` which utilizes FPU registers. By commenting out the lines:

    sprintf( cBuf, "%1.3f", 1.234 );

and 

    configASSERT( strcmp( cBuf, "1.234" ) == 0 );

the demo has been running fine for nearly 45 minutes.

Whereas `FreeRTOS_IRQ_Handler` [saves (and restores) FPU context](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/ddd50d9a80fd80e2bca7e93d49baf5c71986cbbb/portable/GCC/ARM_CR5/portASM.S#L252) via the `[portSAVE_CONTEXT](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/ddd50d9a80fd80e2bca7e93d49baf5c71986cbbb/portable/GCC/ARM_CR5/portASM.S#L54)` macro, `FreeRTOS_Tick_Handler` does not.

**Target**
- Development board: Xilinx ZCU104
- IDE and version: Vitis 2022.2
- Toolchain and version: armr5-none-eabi-gcc 11.2.0

**Host**
- Host OS: Linux Mint (Ubuntu-based)
- Version: 19.3

**To Reproduce**
- Run the FreeRTOS demo on a Xilinx ZCU104 development board. It should fail after less than a minute or so.

**Screenshots**
[image]

[image]

chris-laplante, 2023-03-20:
On second thought, I'm not sure I understand precisely what is going on here. `_svfprintf_r` vpush'es and vpop's the d0-d15 FPU registers it is using. So I'm not sure if `_svfprintf_r` can be blamed here. I don't see `_svfprintf_r` save the fpscr register (which portSAVE_CONTEXT does), however. Not sure if that's important.

chris-laplante, 2023-03-20:
Ah, `_dtoa_r` is the real culprit (which `_svfprintf_r` calls). `_dtoa_r` uses a bunch of FPU registers which it doesn't bother to save or restore.

chris-laplante, 2023-03-20:
Seems like in Cortex R5 the calling convention is that d0-d7 do not need to be preserved across subroutine calls, whereas d8-d15 do. This explains why the compiler only emits code to vpush/vpop d8-d15 but not d0-d7. So d0-d7 need to be saved by the tick handler.

aggarg, 2023-03-21:
As per [this comment](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/GCC/ARM_CR5/portASM.S#L271), the application needs to supply `vApplicationFPUSafeIRQHandler` if it wants FPU registers to be saved on interrupt entry. Would you please try renaming [this function](https://github.com/FreeRTOS/FreeRTOS/blob/main/FreeRTOS/Demo/CORTEX_R5_UltraScale_MPSoC/RTOSDemo_R5/src/FreeRTOS_tick_config.c#L112) to `vApplicationFPUSafeIRQHandler`?

aggarg, 2023-03-21:
Would you double check because I see that here - https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/main/portable/GCC/ARM_CR5/portASM.S#L299. Or am I looking at the wrong port?

aggarg, 2023-03-21:
Thanks. Possible for you to verify with the latest FreeRTOS source?

chris-laplante, 2023-03-24:
> Thanks. Possible for you to verify with the latest FreeRTOS source?

I'll give it a try

Skptak, 2023-04-15:
Hey @chris-laplante, it's been a while since you opened this bug. I was wondering if you were still facing an issue with this or if this was something that you were able to resolve by using the newest version of the FreeRTOS Port in our repository?

chris-laplante, 2023-04-17:
Hi @Skptak, sorry I have not had time recently to test. I plan on testing it later this week.

Skptak, 2023-05-15:
Hey @chris-laplante, I wanted to reach out and see if this was something you were still having an issue with?

chris-laplante, 2023-05-15:
Sorry again for the delay. Go ahead and close this, and I'll reopen in the future if necessary. Haven't had a chance to verify this but don't want to leave the issue hanging open.

closed 2023-05-15 by Skptak