FreeRTOS/FreeRTOS #1014: [BUG]

ZefengWang, 2023-05-12:
**Describe the bug**
FreeRTOS/FreeRTOS/Demo/Posix_GCC demo runs crash

**Target**
- Development board: [x86_64 machine, cpu: amd 7950x]
- Instruction Set Architecture: [x86_64]
- IDE and version: [gcc 11.3.0]
- Toolchain and version: [x86_64-linux-gnu-gcc-11]

**Host**
- Host OS: [Ubuntu 22.04 LTS]
- Version: [jammy]

**To Reproduce**
- Use project Posix_GCC and configure with main_full
- Run on `make clean&&make&&./build/posix_demo` and could observe 
```
[WARN] pthread_attr_setstack failed with return value: 22. Default stack will be used.
[WARN] Increase the stack size to PTHREAD_STACK_MIN.

Trace started.
The trace will be dumped to disk if a call to configASSERT() fails.
Starting full demo
*** stack smashing detected ***: terminated
已放弃

```

**Expected behavior**
A concise description of what you expected to happen.
```
[WARN] pthread_attr_setstack failed with return value: 22. Default stack will be used.
[WARN] Increase the stack size to PTHREAD_STACK_MIN.

Trace started.
The trace will be dumped to disk if a call to configASSERT() fails.
Starting full demo
OK: No errors - tick count 10000 
```
**Screenshots**
[image]

**Additional context**
Add any other context about the problem here.
FreeRTOSv202212.01/FreeRTOS-Plus/Source/FreeRTOS-Plus-Trace/trcSnapshotRecorder.c
in function prvReportStackUsage, there is a variable named unusedStackSpace, its type should be TraceUnsignedBaseType_t rather than uint32_t.
if in different paltform, the length of  TraceUnsignedBaseType_t is not the same. this cause the demo crash.

moninom1, 2023-05-14:
Hi @ZefengWang , 
Thank you for bringing this to our attention! We will be looking further into this issue as soon as we can, thank you for your patience!”

moninom1, 2023-05-15:
Hi @ZefengWang 

FreeRTOSv202212.01 we have fixed similar issue on the [FreeRTOS tip](https://github.com/FreeRTOS/FreeRTOS).
Can you try with FreeRTOS tip and let us know if you are still facing the issue.

tony-josi-aws, 2023-05-26:
Closing this issue for now, please feel free to reopen this if you sill face the issue.

closed 2023-05-26 by tony-josi-aws