FreeRTOS/FreeRTOS #1077: [BUG] va_arg process error

llxiaoyuan, 2023-08-21:
Crush On 64bit exe

https://github.com/FreeRTOS/FreeRTOS/blob/38a57333ed33cbe889efca3b582c22b522cf66b0/FreeRTOS/Demo/PIC32MX_MPLAB/printf-stdarg.c#L146 

to fix:
register char* s = (char*)va_arg(args, char*);

Skptak, 2023-08-24:
Hey @llxiaoyuan, not quite sure I'm following what the issue is here?
You say this a problem on a 64 bit EXE but this demo is for the [PIC32MX](https://www.microchip.com/en-us/products/microcontrollers-and-microprocessors/32-bit-mcus/pic32-32-bit-mcus/pic32mx) family of microcontrollers, which I believe are all 32 bit microcontrollers?
Can you provide more information about what you mean by "Crush on 64 bit exe" please?

kar-rahul-aws, 2023-08-31:
Hi @llxiaoyuan 
Can you provide more information on this issue as suggested by @Skptak

llxiaoyuan, 2023-09-01:
i use this code in windows x64，char * is 8 byte in x64，char * is 4 byte in x86，int is 4 byte in x64 and x86，so you can change int to char* for compatibility

Skptak, 2023-11-20:
Hey @llxiaoyuan, this is unfortunately not a bug with this file.
This function is intended to be built and run on the 32 bit microcontroller. Changing this type from an int to a character pointer would be an incorrect cast of this data type. 
If you are using this code on windows x64 you should modify the file as needed for your use case, and then submit a PR with your new project
Thanks for raising this issue, if you disagree with this please bring it up in the [FreeRTOS forums](https://forums.freertos.org/) so the community can discuss it!

closed 2023-11-20 by Skptak