FreeRTOS/FreeRTOS #1080: [BUG] FULL_DEMO of CORTEX_M3_MPS2_QEMU_GCC doesn't work sometimes

anolivetree, 2023-08-31:
**Describe the bug**
FULL_DEMO of CORTEX_M3_MPS2_QEMU_GCC doesn't work sometimes

**Target**
- CORTEX_M3_MPS2_QEMU_GCC
- QEMU

**Host**
- QEMU on Ubuntu on x86_64
- QEMU on WSL

**To Reproduce**
$ cd Demo/CORTEX_M3_MPS2_QEMU_GCC/
$ make FULL_DEMO=1
$ qemu-system-arm  -machine mps2-an385 -monitor null -semihosting --semihosting-config enable=on,target=native -kernel ./build/RTOSDemo.axf -serial stdio -nographic
and wait for 10 seconds.

**Expected behavior**
FULL_DEMO app should print "OK: No errors", but it sometimes stop with the error message ""Error: Death "

CORTEX_M3_MPS2_QEMU_GCC/main_full.c has a "TimerMode" task which runs about 950ms and vTaskDelete itself. uxCurrentNumberOfTasks is decremented in idle task. 
vCreateTasks in Common/Minimal/death.c sleeps for 1000ms in order to wait for other tasks to be created and set uxTasksRunningAtStart.

When uxCurrentNumberOfTasks is decremented after uxTasksRunningAtStart is set, 
FULL_DEMO app stop with the message "Error: Death - tick count 10000"

kar-rahul-aws, 2023-09-01:
Hi @anolivetree 
Thank you for reporting the issue. We are looking to reproduce the issue and will start working on the fix.

kar-rahul-aws, 2023-09-04:
Hi @anolivetree 
We were able to reproduce the issue, and have created a patch for solving the same.
Can you please confirm, if the demo works for you?

anolivetree, 2023-09-05:
@kar-rahul-aws 
It works on my PC without any problems. Thank you for the fix!

kar-rahul-aws, 2023-09-05:
@anolivetree 
Thank you for the confirmation. Closing this issue.

closed 2023-09-05 by kar-rahul-aws