FreeRTOS/FreeRTOS #1088: Task suspending itself can remove itself from termination list

panduit-mik, 2023-09-11:
**Describe the bug**
Race condition on system with multiple processors. If  TaskA requests to vTaskDelete(taskB) and taskB is currently blocked on taskENTER_CRITICAL() in vTaskSuspend(0), taskB will remove itself from the termination list and put itself on the suspendedTaskList thus never actually getting deleted when IDLE comes around.

**Target**
- Development board: custom
- Instruction Set Architecture: xtensa-esp32
- IDE and version: n/a
- Toolchain and version: xtensa-esp32-elf-gcc

**Host**
- Host OS: WSL
- Version: 1

**To Reproduce**
Seen using pthread implementation on ESP-IDF v4.4 

pthread_join() uses this excerpt
```
        if (wait) {
            xTaskNotifyWait(0, 0, NULL, portMAX_DELAY);
            if (xSemaphoreTake(s_threads_mux, portMAX_DELAY) != pdTRUE) {
                assert(false && "Failed to lock threads list!");
            }
            child_task_retval = pthread->retval;
            pthread_delete(pthread);
            xSemaphoreGive(s_threads_mux);
        }
        vTaskDelete(handle);

```

pthread_exit() uses this sequence:
```
    xTaskNotify(pthread->join_task, 0, eNoAction);
    xSemaphoreGive(s_threads_mux);
    vTaskSuspend(NULL);
```

If the spawned thread (taskB) calls pthread_exit(), the pthread_join() task (taskA) wakes up and attempts to delete the joined thread. If taskB is on one core and taskA is on the other, it is possible that taskB is at taskENTER_CRITICAL() in vTaskSuspend() when taskA via vTaskDelete()  puts taskB on the xTasksWaitingTermination list. When taskB gets access to the critical region, it calls uxListRemove() thinking it is removing the task from the ready/delayed list when it actually ends up removing itself from the xTasksWaitingTermination list.
taskB is then put on xSuspendedTaskList and becomes a zombie thread from the applications point of view.

**Expected behavior**
Task should be deleted when requested. vTaskSuspend() needs to verify that the task being suspended is not on the termination list. If it is, then the code which removes it from ready and puts it on the suspended list needs to be skipped.

rawalexe, 2023-09-12:
Thank you so much for reporting the bug, I'll let the team know and someone can have a look at the problem.

panduit-mik, 2023-09-18:
On an older version of FreeRTOS, this fixes the issue:
`diff --git a/components/freertos/tasks.c b/components/freertos/tasks.c
index 093b3898bd..7b762cc832 100644
--- a/components/freertos/tasks.c
+++ b/components/freertos/tasks.c
@@ -1965,6 +1965,20 @@ static void prvAddNewTaskToReadyList( TCB_t * pxNewTCB,
 
             traceTASK_SUSPEND( pxTCB );
 
+            /* Check if the task is already marked for deletion.
+             * This can happen if this task was waiting to enter the
+             * critical section on one core to suspend itself
+             * when a different task on the other core made the request
+             * to delete it */
+            if (listLIST_ITEM_CONTAINER( &(pxTCB->xStateListItem) ) == &xTasksWaitingTermination)
+            {
+                /* Task was marked for deletion (probably from the other core)
+                 * Do not put it in suspended list. We need to let Idle
+                 * on this core clean it up */
+            }
+            else
+            {
+
                 /* Remove task from the ready/delayed list and place in the
                  * suspended list. */
                 if( uxListRemove( &( pxTCB->xStateListItem ) ) == ( UBaseType_t ) 0 )
@@ -1987,7 +2001,6 @@ static void prvAddNewTaskToReadyList( TCB_t * pxNewTCB,
                 }
 
                 vListInsertEnd( &xSuspendedTaskList, &( pxTCB->xStateListItem ) );
-            curTCB = pxCurrentTCB[ xPortGetCoreID() ];
 
                 #if ( configUSE_TASK_NOTIFICATIONS == 1 )
                     {
@@ -2005,6 +2018,8 @@ static void prvAddNewTaskToReadyList( TCB_t * pxNewTCB,
                     }
                 #endif /* if ( configUSE_TASK_NOTIFICATIONS == 1 ) */
             }
+            curTCB = pxCurrentTCB[ xPortGetCoreID() ];
+        }
         taskEXIT_CRITICAL();
 
         if( xSchedulerRunning != pdFALSE )

`

rawalexe, 2023-09-19:
Hello @panduit-mik ,
What version of the kernel are you using?

panduit-mik, 2023-09-20:
The version currently in use for ESP-IDF v4.4.5  https://github.com/espressif/esp-idf/releases/tag/v4.4.5

chinglee-iot, 2023-09-21:
Hi @panduit-mik

You probably are using Espressif maintained FreeRTOS distribution in esp-idf v4.4.5.
FreeRTOS SMP are included in esp-idf v5.0 as a configuration.
_In menuconfig
(Top) → Component config → FreeRTOS → Kernel
Espressif IoT Development Framework Configuration                                                                      
[*] Run the Amazon SMP FreeRTOS kernel instead (FEATURE UNDER DEVELOPMENT)_

> If TaskA requests to vTaskDelete(taskB) and taskB is currently blocked on taskENTER_CRITICAL() in vTaskSuspend(0), taskB will remove itself from the termination list and put itself on the suspendedTaskList thus never actually getting deleted when IDLE comes around.

The problem you mentioned is addressed in FreeRTOS SMP.
When taskA deletes taskB which is blocked on taskENTER_CRITICAL(), it will set taskB's run state to `taskTASK_SCHEDULED_TO_YIELD` and requests taskB to yield. If a task's run state is changed before entering critical section, it will enable the interrupt and retry entering the critical section later. Enabling interrupt should cause the core to handle the pending yield request from taskA immediately. Therefore, taskB will be preempted by other task and vTaskSuspend( 0 ) won't be executed. Later Idle task can delete taskB. You can reference [prvCheckForRunStateChange ](https://github.com/FreeRTOS/FreeRTOS-Kernel/blob/d43062ba7862f6c354ca414b411958ce12a7187f/tasks.c#L725) function for more details.

chinglee-iot, 2023-09-22:
@panduit-mik 
We would like to suggest you to use esp-idf version after v5.0 and enable FREERTOS_SMP feature for this problem.
This issue will be closed for now. If you have further question, feel free to reopen this issue.

closed 2023-09-22 by chinglee-iot