FreeRTOS/FreeRTOS #1206: [BUG] Wrong conditions in mqtt_pkcs11_demo_helpers.c

KeitaKashima, 2024-04-11:
**Describe the bug**

In mqtt_pkcs11_demo_helpers.c, there would be typo conditions L797 and L805.

I think it should check **NOT pd FAIL**, like`if( xReturnStatus != pdFAIL )` but it performs `if( xReturnStatus == pdFAIL )`.

https://github.com/FreeRTOS/FreeRTOS/blob/eb97c5d325c58cb7d67b34aa735350d8d684120d/FreeRTOS-Plus/Demo/AWS/Mqtt_Demo_Helpers/mqtt_pkcs11_demo_helpers.c#L797

[image]

It seems that the demo is working fine.

Because this `xMqttSessionEstablished` flag is only used when do the MQTT Disconnect.

And when the flag is true, the device sent MQTT DISCONNECT packet.

But after that, the connection is disconnected immidiately. So, it would not affect the demo.

**Target**

Below is my target, but it is happened other devices.

- Development board: [CK-RX65N](https://www.renesas.com/products/microcontrollers-microprocessors/rx-32-bit-performance-efficiency-mcus/ck-rx65n-cloud-kit-based-rx65n-mcu-group) (Renesas RX family)
- Instruction Set Architecture: RX65N
- IDE and version: e2studio 2024-01
- Toolchain and version: CC-RX V3.

**Host**
- Host OS:  Win10

**To Reproduce**
- Use project  : [URL RX65N AWS FreeRTOS v202210 IoT repo](https://github.com/renesas/iot-reference-rx/tree/v202210.01-LTS-rx-1.2.1)
- Run on :  CK-RX65N

**Expected behavior**

Change to `if( xReturnStatus != pdFAIL )` .

xuelix, 2024-04-17:
Thanks for reporting. We'll fix it.

kar-rahul-aws, 2024-05-30:
Closing this issue. since the issue is fixed in this [PR](https://github.com/FreeRTOS/FreeRTOS/pull/1210)

closed 2024-05-30 by kar-rahul-aws

chungkyufum, 2025-08-22:
L805 still not fixed.

xuan97z1, 2026-01-20:
Greetings @xuelix , @kar-rahul-aws ,  
Recently, my team have been developing a Fleet Provisioning application which runs on the CK-RX65N v2 target board,  
We still encountered a bug related to the wrong condition reported above.  
[image]  
  
The original poster raised 2 erroneous conditions (L#797, L#805).  
However, the PR only fixed the first mistake.    
  
The [second mistake in L#805](https://github.com/FreeRTOS/FreeRTOS/blob/main/FreeRTOS-Plus/Demo/AWS/Mqtt_Demo_Helpers/mqtt_pkcs11_demo_helpers.c#L810) still remains.  
    
**Request**

1. As a temporary workaround, we increased the `MAX_OUTGOING_PUBLISHES` to `3` to allow more recovery attempts.  
    Is this acceptable?
2. Could you consider to reopen this issue and continue the discussion?    

Thank you.