FreeRTOS/FreeRTOS #1246: [Feature Request] cancellable xTimerPendFunctionCall

emilfihlman, 2024-08-14:
**Is your feature request related to a problem? Please describe.**
I'm trying to have a function be called after a delay
I want to pass a void * argument to it.
I want to be able to also cancel this call before the requested delay has passed.
I prefer to be able to do anything I want inside the callback, ie not just ISR safe functions calls.

**Describe the solution you'd like**
`
UniqueIdentifierType_t xTimerPendFunctionCallCancellable(
      PendedFunction_t xFunctionToPend,
      void *pvParameter1,
      uint32_t ulParameter2,
      TickType_t xTicksToWait
);
`
ie same arguments as,xTimerPendFunctionCall, just different return type
`
BaseType_t xTimerPendFunctionCallCancellable( UniqueIdentifierType_t id );
`

**Describe alternatives you've considered**
I couldn't find a direct way to do this as
- xTimerPendFunctionCall is seemingly non cancellable
- xTimerCreate and other timer functions is basically cancellable as one can stop the timer and resetting it etc, but it does not allow passing an argument, or if it does with the pvTimerID, the documentation is unclear on what I can I do with it
- xTaskCreate doesn't work directly because as far as I know there is no guarantee that handle values aren't reused

Of course this is achievable with additional work by building a dedicated handler etc, but it would be nice if one could do it directly without much hassle as it's already really close

**How many devices will this feature impact?**
I think this would be very usable for most of the devices using freeRTOS

**What are your project timelines?**
Would be nice if support for this would be quick

xuelix, 2024-08-20:
Thanks for the suggestion. I'll have the team to review the request first.

aggarg, 2024-08-21:
> but it does not allow passing an argument, or if it does with the pvTimerID, the documentation is unclear on what I can I do with it

This is not correct. You can pass a `void*` at the time of timer create - 
```c
TimerHandle_t xTimerCreate( const char * const pcTimerName,
                            const TickType_t xTimerPeriod,
                            const UBaseType_t uxAutoReload,
                            void * const pvTimerID, <-- This is the parameter I am talking about.
                            TimerCallbackFunction_t pxCallbackFunction );
```

You can obtain this value in the timer callback using `pvTimerGetTimerID`:

```c
void TimerCallbackFunction( TimerHandle_t pxExpiredTimer )
{
    void * pUserData;

    pUserData = pvTimerGetTimerID( pxExpiredTimer );

    /* Use this pUserData. */
}
```
Is this what you are looking for?

emilfihlman, 2024-08-21:
@aggarg

The documentation in https://www.freertos.org/Documentation/02-Kernel/04-API-references/11-Software-timers/01-xTimerCreate states

> pvTimerID
> 
> An identifier that is assigned to the timer being created. Typically this would be used in the timer callback function to identify which timer expired when the same callback function is assigned to more than one timer, or together with the [vTimerSetTimerID()](https://www.freertos.org/Documentation/02-Kernel/04-API-references/11-Software-timers/15-vTimerSetTimerID) and [pvTimerGetTimerID()](https://www.freertos.org/Documentation/02-Kernel/04-API-references/11-Software-timers/13-pvTimerGetTimerID) API functions to save a value between calls to the timer's callback function.

What exactly are the guarantees we have for this argument? To me it seems based on this documentation and the used name that passing void *user with this requires careful hand holding and is not at all straightforward.

If the point of the argument is exactly to be able to pass void *user and that it doesn't require careful orchestration then I propose the documentation is changed to reflect on this.

Thank you in either case!

emilfihlman, 2024-08-23:
@xuelix @aggarg could you confirm if there are no gotchas here or stuff to keep in mind if I want to use this?

aggarg, 2024-08-26:
There is no gotchas and you can store and retrieve any `void*` as I described. The documentation only describes one possible use case but you are free to use it whichever way you want.

aggarg, 2024-08-27:
I am closing this issue. Feel free to re-open if you need anything else from us.

closed 2024-08-27 by aggarg