FreeRTOS/FreeRTOS #1256: [DOC] freertos.org directs to midway-auth.amazon.com

emilfihlman, 2024-08-23:
**Describe the issue**
freertos.org directs to some issue page

**Reference**
freertos.org

**Screenshot**
[image]

**Browser**
- Browser: Chrome Android
- Version: 127.0.6533.103

xuelix, 2024-08-23:
Thanks for reporting the bug, we have recently updated the website. That might be a few links missed update.  If you can paste the exact link that leads to this, that'll be a great help to fix the issue.  Thanks

emilfihlman, 2024-08-23:
> Thanks for reporting the bug, we have recently updated the website. That might be a few links missed update. If you can paste the exact link that leads to this, that'll be a great help to fix the issue. Thanks

It would be directly the link https://freertos.org and any link

emilfihlman, 2024-08-23:
But it seems to work now

AniruddhaKanhere, 2024-08-23:
Hello @emilfihlman,
Thank you for raising the bug report.

We were deploying some fixes to the website. The deployment failed though and caused the cloudfront distribution to be this was. We have since rolled it back. You should be able to access the website now.

Let me know if you want to keep this open or shall we close the ticket?

aggarg, 2024-08-26:
As the redirection is working now, I am closing this ticket. Feel free to reopen if you need anything from us.

closed 2024-08-26 by aggarg