FreeRTOS/FreeRTOS #1264: [BUG] transport interface for wolfSSL

rickou, 2024-09-13:
**Describe the bug**
transport_wolfSSL.c include application include file...

it include "user_settings.h" and "demo_config.h"

in wolfSSL, the user_settings.h file is included only when WOLFSSL_USER_SETTINGS is defined

regarding demo_config.h... it has nothing to do in this file... (same for the related code) this could keep the interface as a lib without user modification in it.

am i right in my expectation ?

aggarg, 2024-09-15:
> this could keep the interface as a lib without user modification in it.

This is not a library but an implementation of transport interface which should be used as a reference. Feel free to update this file as per your application needs.

aggarg, 2024-09-17:
I am closing this issue. Feel free to reopen if you need anything from us.

closed 2024-09-17 by aggarg